thonny/thonny #2610: No connection to ESP8266 (D1 mini, 4 MB ROM, WeMos) with Thonny under Linux

HeinzEgger, 2023-01-12:
I have just started programming with MicroPython. I could program using Thonny under Windows 10. But my favorite OS is Linux Ubuntu (22.04.1 LTS). I use Thonny 4.0.1 under Linux. But I cannot connect to the ESP8266 (WeMos D1 mini, 4 MB ROM).
In the terminal I can see, that the hardware, is recognized:

heinz@heinz-HP-ENVY-Notebook:~$ lsmod | grep ch34
ch341 24576 0
usbserial 57344 1 ch341 _=> serial chip on my D1 mini_

and sudo demsg shows:

[ 4201.030065] usb 1-1.4.1: new full-speed USB device number 14 using xhci_hcd
[ 4201.131948] usb 1-1.4.1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.54
[ 4201.131970] usb 1-1.4.1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 4201.131981] usb 1-1.4.1: Product: USB2.0-Ser!
[ 4201.233594] usbcore: registered new interface driver usbserial_generic
[ 4201.233612] usbserial: USB Serial support registered for generic
[ 4201.244829] usbcore: registered new interface driver ch341
[ 4201.244859] usbserial: USB Serial support registered for ch341-uart
[ 4201.244880] ch341 1-1.4.1:1.0: ch341-uart converter detected
[ 4201.245577] ch341-uart ttyUSB0: break control not supported, using simulated break
[ 4201.247507] usb 1-1.4.1: ch341-uart converter now attached to ttyUSB0
[ 4201.274377] input: PC Speaker as /devices/platform/pcspkr/input/input22
[ 4201.797396] input: BRLTTY 6.4 Linux Screen Driver Keyboard as /devices/virtual/input/input23
[ 4201.918566] usb 1-1.4.1: usbfs: interface 0 claimed by ch341 while 'brltty' sets config #1
[ 4201.919523] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 4201.919556] ch341 1-1.4.1:1.0: device disconnected

(although I do not understand why the ttUSB0 is disconnected)

Has anyone a hint, what is going wrong and what I could do?
Thanks
Heinz

aivarannamaa, 2023-01-14:
As it doesn't seem to be specific to Thonny, I think you have more chances with https://github.com/orgs/micropython/discussions

aivarannamaa, 2023-01-14:
Oh, I now noticed `brltty` in your log. This reminded me of another Thonny user struggling with this, see #2360

closed 2023-01-14 by aivarannamaa