thonny/thonny #2613: save as

AakeJon49, 2023-01-13:
When I try to Save As, it is not possible to paste the file name as I have copied. Nothing happen when I ctrl+V

aivarannamaa, 2023-01-14:
Thank you for the report!

Which OS and which Thonny version are you using?

AakeJon49, 2023-01-14:
I use Ubuntu 2022.04.1 LTS and Thonny 4.0.1
Besides this I love Thonny, Thanks!!!

Den lör 14 jan. 2023 kl 10:09 skrev Aivar Annamaa ***@***.***

>:

> Thank you for the report!
>
> Which OS and which Thonny version are you using?

aivarannamaa, 2023-01-14:
I could now reproduce this in RPiOS, but unfortunately I don't have a good solution. It looks like the file dialogs provided by `zenity` don't get along with Tkinter's (Thonny's UI toolkit) approach to clipboard.

You could try using Tk's file dialog (Tools => Options => General => Use Tk file dialogs instead of zenity), but they have their own downsides.

Anyway, I'm closing this issue, as I can't do anything. Let's hope next versions of Tk will fix this.

closed 2023-01-14 by aivarannamaa

AakeJon49, 2023-01-14:
Ok, thanks.

Den lör 14 jan. 2023 kl 18:03 skrev Aivar Annamaa ***@***.***

>:

> Closed #2613 <https://github.com/thonny/thonny/issues/2613> as not
> planned.
>
> <https://github.com/thonny/thonny/issues/2613#event-8227230062>, or
> unsubscribe