thonny/thonny #2614: Errors when I open a .json file

vpatron, 2023-01-14:
I have a file on the board called `config.json`. I get the error below. If I rename it to config.txt it loads fine. Even loading an empty file called `test.json` causes an error.

This started happening in v3.3.14. Earlier versions were fine with opening .json files.

It shows this initial dialog box:
```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'read_file' (thonny.plugins.micropython.mp_back.ManagementError: Command output was not empty).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
```
When I click close on that, it shows another dialog called `Internal Error` with this info:

```
Traceback (most recent call last):
  File "/home/vince/apps/thonny/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/home/vince/apps/thonny/lib/python3.10/site-packages/thonny/workbench.py", line 2027, in toolbar_handler
    handler(*args)
  File "/home/vince/apps/thonny/lib/python3.10/site-packages/thonny/editors.py", line 891, in _cmd_open_file
    self.show_file(path, propose_dialog=False)
  File "/home/vince/apps/thonny/lib/python3.10/site-packages/thonny/editors.py", line 1043, in show_file
    editor = self.get_editor(filename, True)
  File "/home/vince/apps/thonny/lib/python3.10/site-packages/thonny/editors.py", line 1134, in get_editor
    return self._open_file(filename_or_id)
  File "/home/vince/apps/thonny/lib/python3.10/site-packages/thonny/editors.py", line 1116, in _open_file
    if editor._load_file(filename):
  File "/home/vince/apps/thonny/lib/python3.10/site-packages/thonny/editors.py", line 206, in _load_file
    result = self._load_remote_file(filename)
  File "/home/vince/apps/thonny/lib/python3.10/site-packages/thonny/editors.py", line 265, in _load_remote_file
    raise RuntimeError(response["error"])
RuntimeError: Backend terminated
```

Thonny is awesome. But this bug is kind of annoying.

aivarannamaa, 2023-01-14:
Thank you for the report!

It could be helpful if you posted the backend.log. First reproduce the error and then, without reconnecting select "Tools => Open Thonny data folder" and find the backend.log.

aivarannamaa, 2023-01-14:
One more thing -- in case you are still using 3.3.14, please try 4.0.1 first!

aivarannamaa, 2023-01-28:
Please re-open if the problem persists in 4.0.2

closed 2023-01-28 by aivarannamaa