thonny/thonny #2616: ModuleNotFoundError: No module named 'thonny.plugins.micropython.bare_metal_backend'

HeinzEgger, 2023-01-15:
This is a follow-up to closed issue: [#2610](https://github.com/thonny/thonny/issues/2610)
I followed the steps of issue [#2360](https://github.com/thonny/thonny/issues/2360) and had no success. Today I started my machine and removed Thonny, reinstalled it with your file thonny-4.0.1.bash and well, I can connect to the ESP8266. But there is always an error message:
_Traceback (most recent call last):
  File "/home/heinz/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py", line 5, in <module>
    from thonny.plugins.micropython.bare_metal_backend import (
ModuleNotFoundError: No module named 'thonny.plugins.micropython.bare_metal_backend'

Process ended with exit code 1._
[image]

I looked through all the mentioned paths in the error message, and did not find anything missing. See attached screenshot.
What is going wrong?

aivarannamaa, 2023-01-16:
Thank you for the report!

Please reproduce this issue and then, without doing anything else, select "Tools => Options => Open Thonny data folder" and post the backend.log you find in this folder.

HeinzEgger, 2023-01-16:
I have done so. But the file backend.log does not show anything. There was an old file of the same name, dating back to the 12. December 2022 (attached). But I think this has no relevance. I created a new backend.log file but it stayed empty. The frontend.log contains some information (attached). Have I to change Thonny's settings? Mine are as shown in the screenshot (attached). Also the export of the protocol files does not show any results. The zip file is empty. 
[image]
[20221213-backend.log](https://github.com/thonny/thonny/files/10427234/20221213-backend.log)
[frontend.log](https://github.com/thonny/thonny/files/10427236/frontend.log)

aivarannamaa, 2023-01-18:
I'm afraid you have both new and old Thonny versions in your system. 

Do you have `thonny` in `~/.local/lib/python3.10/site-packages` ? If yes, then please delete it from there and try again.

HeinzEgger, 2023-01-18:
Many thanks! Yes, this was the problem. I removed the directory thonny in ~/.local/lib/python3.10/site-packages and the new Thonny-installation works perfektly! I am so happy! And it is a lot faster than under windows!

aivarannamaa, 2023-04-08:
In the next Thonny version Thonny should not be confused by other installations anymore.

closed 2023-04-08 by aivarannamaa

aivarannamaa, 2023-08-11:
More conservative approach implemented in d2a96d04 because of #2851 and #2864.