thonny/thonny #2617: Backend problem Nucleo L476RG

moon-66FB, 2023-01-15:
I have a problem on my Mac with MicroPython and the Nucleo Board L476RG. When I connect it this  error message occurs 

PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ProtocolError: Unexpected read during raw paste).

See Thonny's backend.log for more info.

You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.Process ended with exit code 0.

If I connect the board on Win11 no problems occur !!!

Other Nucleo Board F-401RE works without a problem !!!

Have anybody the same problem or a resolution for this ???

Regards Frank

aivarannamaa, 2023-01-16:
Thanks for the report!

I don't know what may cause this, but you could try a work-around: https://github.com/thonny/thonny/wiki/MicroPython#advanced-configuration (see the `submit_mode` block)

moon-66FB, 2023-01-17:
Thanks for your reply,
I check this but it seems OK (I compare it with the .ini from Win11)
I have copied the backend logs from the working F-401 Nucleo board and the not working L-476 Nucleo board.
Hope you can figure out why this failure occurs.

[backend-F401.log.odt](https://github.com/thonny/thonny/files/10433434/backend-F401.log.odt)
[backend-476.log.odt](https://github.com/thonny/thonny/files/10433436/backend-476.log.odt)

aivarannamaa, 2023-01-17:
In this case the logs don't add much, as the message "Unexpected read during raw paste" already says MicroPython (or USB driver) is not working as expected.

Did you try another `submit_mode`, e.g. `submit_mode = paste`?

moon-66FB, 2023-01-18:
Success! But the behavior is a something strange ...

I open the config file and put the submit_mode string on the first row (submit_mode = paste) nothing change, failure !
I check the other possibilities , raw and raw_paste  nothing change, failure ! 
(I test all this possibilities with the other Nucleo F-401 board, works absolutely perfect !)
At last I edit the submit_mode back to "= paste" and do a last test... and it works !!!  ???
I take a look in the backend log and there I read "submit_mode = raw"  and the same in the config file its change
 to " =raw"
However, it works but why now ???
Thanks a lot for your excellent Help !  (Thonny is an amazing  IDE. Before I work with Arduino IDE and with Thonny I do 
my first steps in MicroPython.)

aivarannamaa, 2023-01-18:
At launch Thonny reads configuration.ini into memory and writes (possibly modified) configuration back to file on shutdown. If you edited the file when Thonny was running, then overwriting your edits is expected.

closed 2023-01-28 by aivarannamaa