thonny/thonny #2618: Thonny Shell somehow cancels f-string when it generates a tab character

Stehlampe2020, 2023-01-16:
I have a little simple encrypting script using `^` (xor) on a string's ASCII integers and respective certificate array like eg. `[0, 5, 22, 88, 67, 39, 7]`.   
I then test-wise print out multiple runs in a multi-line f-string.   
The problem is that the encryption generates a tab character (followed by a `M`) and the resulting string ends immediately after the tab character, even though there's some more text following in the f-string expression ad outside the expression.   

[Screen recording of the issue](https://user-images.githubusercontent.com/94976382/212660785-65054325-571b-4d82-8aa5-362d13d1fe2b.webm)   

If you wonder, that's Thonny 4.0.1, I just removed the ukraine button but otherwise this install is unmodified.

encrypt.py:
```python3
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

def openCertfile(filename='l2cert.l2cert'):
    with open(filename, 'rb') as certfile:
        return tuple([char for char in certfile.read()]) # Iterating over a bytestring returns integers.

def loopCert(cert=(0,)):
    """Give back one place of the certificate at a time, enabling infinite looping over the certificate.
If no cert is given, only zeroes will be returned which means that the en-/decoding will do nothing with the input."""
    while True:
        for place in cert:
            yield place

def cryptWithCert(string, cert=(0,)):
    cert_iterator = loopCert(cert)
    return ''.join([chr(ord(char)^next(cert_iterator)) for char in string])

if __name__ == '__main__':
    print(f"""
Encrypted with only zeroes (should not do anything):
{cryptWithCert('Soo secret message!')}

Encrypted with contents of l2cert.l2cert:
{cryptWithCert('Soo secret message!', openCertfile())}

Twice encrypted with contents of l2cert.l2cert (second time should decrypt it):
{cryptWithCert(cryptWithCert('Soo secret message!', openCertfile()), openCertfile())}
""")

```

l2cert.l2cert: (just some random text, if you don't understand it that's not important. The exact value is just important to the en-/decoder)
```none
Hallo Welt! Dies ist ein GROßER TAG, da ich einen guten Weg gefunden habe, EINEN BELIEBIGEN STRING ZU KODIEREN UND ZU DEKODIEREN!!!
```

aivarannamaa, 2023-01-19:
Do I understand correctly, that the problem is Thonny treating some non-printable characters differently from the Gnome Terminal?

This is expected, as Thonny's Shell is not a proper terminal emulator. It can recognize some control character, but definitely not so many as Gnome Terminal.

Stehlampe2020, 2023-01-19:
My problem isn't that it handles non-printable characters differently, my problem is that it cuts off the f-string or even the `print()` function on that character sequence.

aivarannamaa, 2023-01-20:
Can you give me a simple program or the string for reproducing the problem? You can output the string literal by wrapping the string to be printed with `repr` -- `print(repr(...))`

Stehlampe2020, 2023-01-20:
With repr I get `\x1b\x0e\x03L\x1cE4\x17\t\x00\x01M!\x1a\x16\x12G\x0cR`.

aivarannamaa, 2023-01-20:
It turns out the NULL-character (`\x00`) is causing problems (see `print("\x00abc")`). It looks like problem in the UI toolkit (Tk) -- (https://stackoverflow.com/questions/71879883/how-to-stop-tkinter-textbox-from-terminating-string-on-null-character).

I'm closing this issue because the solution is out of my reach and printing out non-printable characters (other than terminal control characters) is not a common use case.

closed 2023-01-20 by Stehlampe2020