thonny/thonny #2619: Inconsistencies with Executing without saving

juanfal, 2023-01-18:
The new "feature" Thonny shows, the fact that it allows the student to execute without saving the file.py and so, without it to belong to a working directory, it is not only dangerous, it's misleading for the student as it takes a file as something belonging to Thonny, not to the file system. This is something that is reasonable with the Console and, even there is not so clear, as you don't know where are you, what is the working directory, something obscure if you are specifically working with files.

The situation is even more inconsistent now, as the new "feature" has been added without changing things like the Thonny Setting:
[image]

I love Thonny, and every year now, for 6 years, it has been the tool not only for our 100 students, but for many colleagues I have shared it with.  But I see this as a step backwards and made it inconsistent and playful

aivarannamaa, 2023-01-18:
> it takes a file as something belonging to Thonny, not to the file system.

I'd say, it takes the editor text as something belonging to Thonny, not to to the file system (at least until the file is not saved). In this case even the launch command line (`%Run -c $EDITOR_CONTENT`) indicates that we are working with program text, not with a file.

As I've said elsewhere, I find it instructive to separate the concerns of composing, running and persisting a program text. With earlier Thonny versions I've received several bug reports about "Thonny pops up the Finder/WindowsExplorer/Files window when I'm trying to run my program" -- the user was thinking about their program and didn't expect the file save dialog in this state. Then I realized they shouldn't have to think about persistence when they are eager to test their program for the first time.

I know about the problems with Thonny crashing and students losing their work, but I believe this needs a different solution (#2570).

Regarding "Change working directory to script directory on Run / Debug" -- I don't find it too confusing together with unsaved programs. Unsaved programs are run with current working directory, so nothing needs to be changed in this case.

As it seems to be an important issue for you, I agree to add a checkbox on "Run & Debug" configuration page, which can be used to turn on the previous behavior. It would be OFF by default (unless you give me more arguments for previous behavior). When it's ON, Thonny would present a Save dialog when user tries to run an unsaved program and would refuse to run the program if the user cancels the Save dialog. Would you like this addition?

aivarannamaa, 2023-01-18:
An analogy just occurred to me ... would you expect (or like) MS Word or Libre Office Writer to refuse printing a document unless you have saved it?

juanfal, 2023-01-18:
That’s more than enough, Aivaar, and I have to say here how much I appreciate your open attitude and constant support. I understand what you said about the obstructive Save dialogue, for me it will be more important to teach students in general to save just after a few keystrokes as Python also is good to teach how to indent, no matter the inconveniences. Thanks!--Juan FC

juanfal, 2023-01-19:
I didn’t read this, and well, sorry, I would like to have a kind of in-person “debate”, could it be called that way?, about this topic, it could be called: teaching good programming habits in our students.  I’m close to my retirement.  I resist to take it, but in less than 10 years I won’t be teaching how to program anymore.  Norms like:

Never ever use globals
Do not print from inside subprograms, except they done for that specific purpose
Do not use for loops when you are going to break them.  Use while instead
Do not ever use break (in Python, at least, in C only in its poor switch control structure)
Do not have more than 1 return at the end of functions
Use 01.hell.py, 02.circleArea.py, 03.anotherThing.py… naming convention
etc

Saving files, could be the first, and I even don’t say it.  I come from a world in which any code crash used to yield to restart the computer, so saving files was more than fingers exercising.

When I realised in class Thonny was able to execute its buffer (without saving), for me it was like a gun in hands of a crazy man.  Students at the age of my students do not understand the previous many norms we have at first exercises.  Programming is an Art (I totally stick here to D.E. Knuth), and as an art it is only commanded with practice and experience.  As teacher, you receive many times pieces of code that do what we ask the students to do in  the examples we provide and for that only reason the students say their code is right.  I’ve learned, with the experience, too, that teaching how to program is more teaching good habits and disciplined thinking than writing lines of code

Thonny Console is a bless.  I ask my students to try everything there.  But writing a file, a program, a .py document, is another thing.  If you make these naive minds to think it is the IDE responsibility to keep their code safe, you’re really spoiling them.

Python makes mandatory to use indentation.  Although I hate it, and I for many other reasons I would use Ruby instead of it, it makes some good on students.  It makes them see the importance of aesthetics.  Python should have more syntax rules for, for example variable naming, avoiding 1-letter long range names, but it also allows you to use global-to-the-context variables, so it seems it was not in Guido’s mind to teach good habits in students.

To finish this improvised dissertation, sorry for its length, let me add, the occasions I see students using the “execution of buffer” feature, with 4 o 5 tabs opened and trying with infinite loops in many of those “files.py” (untitled), to make their code to work, I deeply feel that’s not for students, neither for professionals, I would say, but only for really really experienced and a bit wild programmers which don’t care their work is organised in folders and files.  And I sincerely think we, as teachers, shouldn’t offer that route to our students, but safe techniques to work, saving their time and making them have the correct discipline to enjoy their creativity.

Thanks Aivaar for your excellent app.  This world would be mucho () better with only a few people like you.

Warmest regards

- Juan Falgueras

> 
> 
> An analogy just occurred to me ... would you expect (or like) MS Word or Libre Office Writer to refuse printing a document unless you have saved it?
Y ni siquiera lo digo

aivarannamaa, 2023-02-12:
Thonny 4.0.3 will have option "Allow running unnamed programs" under "Run/Debug" options page (ON by default)

closed 2023-02-12 by aivarannamaa