thonny/thonny #2621: Automatic syncing of clock is confusing

veebch, 2023-01-21:
This can lead to differing behaviour between code running via Thonny and code running automatically when connected to power.

What is the purpose of this feature? Shouldn't the default be not to sync?

aivarannamaa, 2023-01-23:
> I realise that this is for the purposed of timestamps on files etc

Yes, that's the main purpose

> because of it, it is easy to end up with code that works when run via thonny, and crashes when not initiated via thonny.

How so? I imagine, if a program depends on the correct time, then it shouldn't be a surprise when it acts erroneously when the time is not correct.

Did you know that you can turn off automatic syncing?

veebch, 2023-01-23:
I know now :) 

My mental map assumed that Thonny would be a window into what happens when the code runs automatically on a pico. Turns out I was wrong....

Thanks!

closed 2023-01-23 by veebch