thonny/thonny #2622: Show class variables in variable view

jrmoserbaltimore, 2023-01-21:
During debugging, I have code such as:

```python
    def set_state(self,
                   Q: Optional[np.single] = None,
                   q: Optional[np.single] = None,
                   frequency: Optional[np.single] = None,
                   bandpass_z1: Optional[np.single] = None,
                   lowpass_z1: Optional[np.single] = None,
                   highpass: Optional[np.single] = None,
                   bandpass: Optional[np.single] = None,
                   lowpass: Optional[np.single] = None):
        if Q is not None: self._q = np.single(1 / Q)
        if q is not None: self._q = np.single(q)
        if frequency is not None: self._f = self._get_f(frequency)
        if bandpass_z1 is not None: self._bandpass_z1 = np.single(bandpass_z1)
        if lowpass_z1 is not None: self._lowpass_z1 = np.single(lowpass_z1)
        if highpass is not None: self._highpass = np.single(highpass)
        if bandpass is not None: self._bandpass = np.single(bandpass)
        if lowpass is not None: self._lowpass = np.single(lowpass)
```

In the Variables view, I have globals, and I have `self` as `<__main__.DSVF object at 0xDEADBEEF>`, but I have no way to see what value is set to self._q, self._f, and so forth.

It's immensely difficult to track the state of this thing like this.  My code matches the diagram for this filter but it doesn't behave correctly.  I've gone through it several times.  I'm keeping track of what the state must be in my head, but it's probably right?  Would be more useful if I could actually inspect it.

As-is, it might be (a lot) easier for me to do this in an HDL design and debug the hardware implementation instead of the software implementation.  Things like `gtkwave` let me inspect every part of the state at any given point, every variable, every register, everything, so it's relatively easy to track down obscure and complex bugs.  As it stands, Thonny's debugging interface feels closer to just not having a way to inspect state.

jrmoserbaltimore, 2023-01-21:
It seems the object inspector does in fact update if automatic stack view is disabled and you close the Stack tab.

aivarannamaa, 2023-01-26:
Thanks for the notification!

I'll try to make the UX smoother.

aivarannamaa, 2024-05-30:
In the upcoming Thonny 5.0 you can move the Stack pane above the Object Inspector, so they won't interfere with each-other.

closed 2024-05-30 by aivarannamaa