thonny/thonny #2623: Problem with deep sleep

PatLub85, 2023-01-22:
Hi, 
I use an ESP8266, Micropython 1.19.1 and Thonny 4.0.2
I have a small program that works with a DHT22. When I use Deep-sleep, and when the ESP wake up , I have the following message:
>>> PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_globals' (thonny.plugins.micropython.mp_back.ManagementError: Script produced errors).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
My code :
def deep_sleep(msecs):
  rtc = machine.RTC()
  rtc.irq(trigger=rtc.ALARM0, wake=machine.DEEPSLEEP)
  # set RTC.ALARM0 to fire after X milliseconds (waking the device)
  rtc.alarm(rtc.ALARM0, msecs)
  print("Dormir")
  machine.deepsleep()

Do you have a solution for this problem?
Thanks for your help

PatLub85, 2023-01-22:
Sorry:
def deep_sleep(msecs):
rtc = machine.RTC()
rtc.irq(trigger=rtc.ALARM0, wake=machine.DEEPSLEEP)
rtc.alarm(rtc.ALARM0, msecs)
print("Dormir")
machine.deepsleep()

aivarannamaa, 2023-01-28:
Thank you for the report!

How did you run your code? By saving it to `main.py` and rebooting your device or with Thonny's Run command?