thonny/thonny #2624: Backend problem (Could not get raw-paste confirmation)

chechigp, 2023-01-25:
After installing the Thonny last version 4.0.2 from the 4.0.1 a backend error message appears every time I try to communicate with the ESP8266. I'm under Win10.
ESP is freezing and rebooting, showing some data in console and finally the backend error appears:

```
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 30756, room 16 
tail 4
chksum 0xf9
load 0x3ffe8000, len 996, room 4 
tail 0
chksum 0x5f
load 0x3ffe83f0, len 1080, room 8 
tail 0
chksum 0xb6
csum 0xb6
����n쒜p��N|�l$ll c���2r�l�N��o�l ��r�$���$`��r�l�$�l`��{�p����ld sd��n���c#��b|$b���b|�����l$b��N�oN
MicroPython v1.19.1-837-g67fac4ebc on 2023-01-24; ESP module with ESP8266
Type "help()" for more information.
PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ProtocolError: Could not get raw-paste confirmation).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.
```

I tried to uninstall and return to older version 4.01 or 3.xx and the behaviour is the same.
I tried to remove the C:\Users\XXXX\AppData\Roaming\Thonny folder with no luck. Trying something similar as I read here #2616
I tried to change the submit_mode in configuration.ini  as I read here #2617 with no luck
I tried with the last stable micropython firmware with also no luck.
I changed the USB port, cable and the ESP device with the same behavior.
I tried to remove all the windows registers with "thonny" input.

Sometimes the backend.log do not catch the erros as I can see, but now it showed this:

```
12:38:16.447 INFO    thonny: Thonny version: 4.0.2
12:38:16.447 INFO    thonny: cwd: C:\Users\XXXX\AppData\Local\Programs\Thonny
12:38:16.447 INFO    thonny: original argv: ['C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\python.exe', '-u', '-B', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp\\esp_back.py', "{'clean': True, 'port': 'COM3', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
12:38:16.447 INFO    thonny: sys.executable: C:\Users\XXXX\AppData\Local\Programs\Thonny\python.exe
12:38:16.447 INFO    thonny: sys.argv: ['C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp\\esp_back.py', "{'clean': True, 'port': 'COM3', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
12:38:16.447 INFO    thonny: sys.path: ['C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\python310.zip', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\DLLs', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\lib', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages']
12:38:16.447 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
12:38:16.447 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': 'COM3', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
12:38:16.525 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
12:38:16.525 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type EspMicroPythonBackend
12:38:16.525 INFO    thonny: TIME/MODS 0.172 before prepare (+73 modules)
12:38:16.525 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_ctypes', '_datetime', '_heapq', '_lzma', '_queue', '_struct', '_tkinter', 'binascii', 'bz2', 'ctypes', 'ctypes._endian', 'ctypes.wintypes', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'msvcrt', 'pathlib', 'platform', 'posixpath', 'queue', 'serial', 'serial.serialutil', 'serial.serialwin32', 'serial.win32', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
12:38:16.525 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
12:38:16.525 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
12:38:16.557 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
12:38:16.588 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
12:38:16.588 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'OK\r\n'
12:38:16.588 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'MPY: soft reboot\r\n'
12:38:25.027 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
12:38:25.027 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ' ets Jan  8 2013,rst cause:4, boot mode:(3,6)\r\n'
12:38:25.027 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
12:38:25.027 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'wdt reset\r\n'
12:38:25.027 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'load 0x40100000, len 30756, room 16 \r\n'
12:38:25.042 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'tail 4\r\n'
12:38:25.058 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'chksum 0xf9\r\n'
12:38:25.058 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'load 0x3ffe8000, len 996, room 4 \r\n'
12:38:25.058 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'tail 0\r\n'
12:38:25.058 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'chksum 0x5f\r\n'
12:38:25.058 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'load 0x3ffe83f0, len 1080, room 8 \r\n'
12:38:25.058 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'tail 0\r\n'
12:38:25.058 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'chksum 0xb6\r\n'
12:38:25.058 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'csum 0xb6\r\n'
12:38:25.120 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '��\x02�\x12�n��r��o|�\x0c$\x0c$dl \x03\x1c#�2\x03'
[... 33 lines cut ...]
  File "C:\Users\XXXX\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\mp_back.py", line 157, in __init__
    self._prepare_rtc()
  File "C:\Users\XXXX\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\mp_back.py", line 197, in _prepare_rtc
    self._check_sync_time()
  File "C:\Users\XXXX\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\mp_back.py", line 208, in _check_sync_time
    self._sync_time()
  File "C:\Users\XXXX\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 388, in _sync_time
    val = self._evaluate(script)
[... cut ...]
    self._submit_code_via_raw_paste_mode(to_be_sent)
  File "C:\Users\XXXX\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 646, in _submit_code_via_raw_paste_mode
    raise ProtocolError("Could not get raw-paste confirmation")
thonny.plugins.micropython.mp_back.ProtocolError: Could not get raw-paste confirmation

```

So I'm out of ideas, I don't know how to back to life my Thonny :S, any ideas?

Thanks in advance and thanks for your effort with this project :)

aivarannamaa, 2023-01-25:
Thank you for the report!

> I tried to change the submit_mode in configuration.ini as I read here https://github.com/thonny/thonny/issues/2617 with no luck

If you configured Thonny to use another `submit_mode` but still get "Could not get raw-paste confirmation", then there is probably something wrong with your configuration. Can you share the configuration file or the relevant section of it?

chechigp, 2023-01-25:
Thanks for your prompt response!

When I change the `submit_mode` to paste I have another error:
```
Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - reset the device and try again;
  - check connection properties;
  - make sure the device has suitable MicroPython / CircuitPython / firmware;
  - make sure the device is not in bootloader mode.

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 30756, room 16 
tail 4
chksum 0xf9
load 0x3ffe8000, len 996, room 4 
tail 0
chksum 0x5f
load 0x3ffe83f0, len 1080, room 8 
tail 0
chksum 0xb6
csum 0xb6
����o�|r��o|�$dld`b���{��l�o��o�l`��r�$���l ��s�p��$�l ��{�l���ll {l��sl��bc��B|l#���b|����$l#�p~�onN�
MicroPython v1.19.1-837-g67fac4ebc on 2023-01-24; ESP module with ESP8266
Type "help()" for more information.
PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.connection.ReadingTimeoutError: Read bytes: bytearray(b'')).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.
```
As I can see the content of this `bytearray(b''))` changes sometime.
The backend.log shows:
```
18:03:24.443 INFO    thonny: Thonny version: 4.0.2
18:03:24.443 INFO    thonny: cwd: C:\Users\XXXX\AppData\Local\Programs\Thonny
18:03:24.443 INFO    thonny: original argv: ['C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\python.exe', '-u', '-B', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp\\esp_back.py', "{'clean': True, 'port': 'COM3', 'dtr': None, 'rts': None, 'submit_mode': 'paste', 'interrupt_on_connect': True, 'write_block_size': 255, 'write_block_delay': 0.03, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
18:03:24.443 INFO    thonny: sys.executable: C:\Users\XXXX\AppData\Local\Programs\Thonny\python.exe
18:03:24.443 INFO    thonny: sys.argv: ['C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp\\esp_back.py', "{'clean': True, 'port': 'COM3', 'dtr': None, 'rts': None, 'submit_mode': 'paste', 'interrupt_on_connect': True, 'write_block_size': 255, 'write_block_delay': 0.03, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
18:03:24.443 INFO    thonny: sys.path: ['C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\python310.zip', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\DLLs', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\lib', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny', 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages']
18:03:24.443 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
18:03:24.443 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': 'COM3', 'dtr': None, 'rts': None, 'submit_mode': 'paste', 'interrupt_on_connect': True, 'write_block_size': 255, 'write_block_delay': 0.03, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
18:03:24.506 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: paste, write_block_size: 255, write_block_delay: 0.03, read_block_size: 0
18:03:24.506 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type EspMicroPythonBackend
18:03:24.506 INFO    thonny: TIME/MODS 0.172 before prepare (+73 modules)
18:03:24.506 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_ctypes', '_datetime', '_heapq', '_lzma', '_queue', '_struct', '_tkinter', 'binascii', 'bz2', 'ctypes', 'ctypes._endian', 'ctypes.wintypes', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'msvcrt', 'pathlib', 'platform', 'posixpath', 'queue', 'serial', 'serial.serialutil', 'serial.serialwin32', 'serial.win32', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
18:03:24.506 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
18:03:24.506 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
18:03:24.584 INFO    thonny.plugins.micropython.bare_metal_backend: Poking
[... cut ...]
  File "C:\Users\XXXX\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 126, in read_all_expected
    actual = self.read(len(expected), timeout=timeout)
  File "C:\Users\XXXX\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 58, in read
    raise ReadingTimeoutError(read_bytes=self._read_buffer)
thonny.plugins.micropython.connection.ReadingTimeoutError: Read bytes: bytearray(b'')

```

Attached you can find the configuration.ini file.

Thanks!
[configuration.zip](https://github.com/thonny/thonny/files/10501627/configuration.zip)

chechigp, 2023-01-25:
Well I'm not sure that this problem is related to Thonny, maybe something changed in my OS because I tried MuEditor and there is a strange behavior too.
Please freeze this issue and I will reply when I have time to test other IDEs to confirm that work.

Thanks for your time!

aivarannamaa, 2023-01-27:
Thanks for the logs!

It looks like your board is having problems with the `paste` mode as well. I've seen the same happen with some ESP32 devices, which occasionally skip characters in `paste`-mode echo.

The last mode to try is `raw` mode. You could start like this:

```
submit_mode = raw
write_block_size = 127
write_block_delay = 0.01
```

If you don't get good results then try a really conservative one:

```
submit_mode = raw
write_block_size = 30
write_block_delay = 0.05
```

If this works, then try increasing the block size or reducing the delay to find the sweet spot.

chechigp, 2023-02-02:
> Thanks for the logs!
>
> It looks like your board is having problems with the `paste` mode as well. I've seen the same happen with some ESP32 devices, which occasionally skip characters in `paste`-mode [...]

Thanks @aivarannamaa for your reply!
I tested it but didn't work, the behavior is the same in booth cases. Finally I found what happened to me and is not related to a Thonny problem. I will record it here and maybe is helpful for someone :)

After countless test with different computers, OS, cables, thonny versions, micropyhton version and different ESP from the same batch I found that the 5G of the Wifi causes soft reboot in these development boards (Amica NODEMCU with 30pin for ESP8266).
And is not related to a ESP wifi configuration the problem is that the 5G band causes interferences with the board in somehow and it soft reset continuously even if is USB connected or battery powered, even if the board is empty without any script in it.
Just some weeks ago I installed a new mesh wifi system with the 5G band, when I turn off the 5G wifi or shield all the ESP module with aluminum foil connected to earth the USB communications come back to live and work without problems.
I installed the new mesh wifi just in the same time that I updated the Thonny version and that confused me.
With the ESP32 development board (NodeMCU ESP-32S V1.1 101010 with 38pin) I don't have any problem to use and work with it, I bought another batch of ESP8266 from other brand and I will test if it have problems with the 5G band, I hope not.

A curiosity about this problem is that a month ago I gifted a friends a small toy for a his baby with an ESP8266 module inside from this problematic batch and it failed in the same time that arrived to his home, I supposed  that was a bad connection or something and when I tested in my home it come back to work again until I installed the new 5G mesh wifi. So the 5G wifi of my friends also caused interferences to the module.
Now is crystal clear but it take me weeks to understand what was happening!

Thanks for your time and your work with this project!

chechigp, 2023-02-12:
Finally seems that I solved the problem including these lines in my script:
`import network
ap_if = network.WLAN(network.AP_IF)
ap_if.active(False)`

Quite strange because I do not use any of the wifi features and it fails also when the ESP is fresh installed from micropython firmware.
I will try to put an issue in micropython git because It seems a strange bug with some wifi config.

Thanks again and best regards

maaganm-hub, 2023-03-06:
hey i need help with this

`Traceback (most recent call last):
  File "/usr/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/base_file_browser.py", line 638, in <lambda>
    label=tr("Open in Thonny"), command=lambda: self.open_file(selected_path)
  File "/usr/lib/python3/dist-packages/thonny/base_file_browser.py", line 1005, in open_file
    get_workbench().get_editor_notebook().show_remote_file(path)
  File "/usr/lib/python3/dist-packages/thonny/editors.py", line 956, in show_remote_file
    return self.show_file(make_remote_path(target_filename))
  File "/usr/lib/python3/dist-packages/thonny/editors.py", line 943, in show_file
    self.select(editor)
  File "/usr/lib/python3.10/tkinter/ttk.py", line 885, in select
    return self.tk.call(self._w, "select", tab_id)
_tkinter.TclError: .!frame.!automaticpanedwindow.!automaticpanedwindow2.!editornotebook.!editor2 is not managed by .!frame.!automaticpanedwindow.!automaticpanedwindow2.!editornotebook

`

i get this.. thonny worked good and suddenly it wont connect about 95% of the times and never load the files

im using esp32

closed 2023-05-27 by aivarannamaa

Cloudineer, 2025-01-28:
I found this issue after seeing similar error messages. Digging a little I found that plugging my USB to serial into a hub caused it to lose/ miss parts of the response from the Python device.

Plugging the USB adapter directly into the computer solved the issue. Hope this helps someone.