thonny/thonny #2625: Hard coded command names

reda-idtaleb, 2023-01-25:
When the backend is terminated by an exit code it tries to restart by calling the following function located at running.py:
```py
def _should_restart_after_command(self, cmd: CommandToBackend):
        print("CMD name = %s\n\n" % cmd.get("name"))
        if cmd is None or not hasattr(cmd, "name"):
            return False

        return cmd.name in ["Run", "run", "Debug", "debug", "execute_source"]
```
The problem is when developping new plugin(with an other command name) the restart of the backend cannot be done after it was terminated by an exit code. The name of the plugin command is obsviously not contained in the list shown on the code on the return statement.

aivarannamaa, 2023-01-26:
Thanks for this notification!

My actual goal was to allow automatic restarts for commands which were directly invoked by the user, in order to avoid endless loop when an automatic creation of the backend fails.

I'll give it some thought... perhaps I should rather blacklist certain commands instead of whitelisting run, debug etc.

reda-idtaleb, 2023-01-26:
I guess this is so restrictive (when defining a hard codes commands in a list) because in my case, for example, I'm developing a  plugin that runs the doctests and when the students click quickly and several times on the plugin button that raises the exception in the shell and the students don't understand what happens with their thonny. It's so confusing for them.

And I guess there's no way to override this function in the code of my plugin.