thonny/thonny #2628: Cannot install micropython-umqtt.robust on Pico W

WobblyRobbly, 2023-01-27:
Hi
I was having issues installing `umqtt.simple` on my Pico W using the Package Manager, however that was fixed as part of #2524 and is working using Thonny 4.0.2 on Raspberry Pi 4B. 

I have been trying to install `micropython-umqtt.robust` in the same way but get the following error (different to previously with `umqtt.simple`)

```
install micropython-umqtt.robust
Looking in indexes: http://127.0.0.1:36628, https://www.piwheels.org/simple
127.0.0.1 - - [27/Jan/2023 17:09:58] "GET /micropython-umqtt-robust/ HTTP/1.1" 200 -
Collecting micropython-umqtt.robust
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 50708)
Traceback (most recent call last):
  File "/usr/lib/python3.9/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/rob/apps/thonny/lib/python3.9/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 207, in __init__
    super().__init__(request, client_address, server)
  File "/usr/lib/python3.9/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.9/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
    method()
  File "/home/rob/apps/thonny/lib/python3.9/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 214, in do_GET
    self._serve_file(*path.split("/"))
  File "/home/rob/apps/thonny/lib/python3.9/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 248, in _serve_file
    tweaked_bytes = self._tweak_file(dist_name, file_name, original_bytes)
  File "/home/rob/apps/thonny/lib/python3.9/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 306, in _tweak_file
    with in_tar.extractfile(info) as f:
  File "/usr/lib/python3.9/tarfile.py", line 2120, in extractfile
    return self.extractfile(self._find_link_target(tarinfo))
  File "/usr/lib/python3.9/tarfile.py", line 2418, in _find_link_target
    raise KeyError("linkname %r not found" % linkname)
KeyError: "linkname 'micropython-umqtt.robust-1.0/umqtt/../../umqtt.simple/umqtt/simple.py' not found"
----------------------------------------
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /micropython-umqtt-robust/micropython-umqtt.robust-1.0.tar.gz/
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 50720)
Traceback (most recent call last):
  File "/usr/lib/python3.9/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/rob/apps/thonny/lib/python3.9/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 207, in __init__
    super().__init__(request, client_address, server)
  File "/usr/lib/python3.9/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.9/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
    method()
  File "/home/rob/apps/thonny/lib/python3.9/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 214, in do_GET
    self._serve_file(*path.split("/"))
  File "/home/rob/apps/thonny/lib/python3.9/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 248, in _serve_file
    tweaked_bytes = self._tweak_file(dist_name, file_name, original_bytes)
  File "/home/rob/apps/thonny/lib/python3.9/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 306, in _tweak_file
    with in_tar.extractfile(info) as f:
  File "/usr/lib/python3.9/tarfile.py", line 2120, in extractfile
    return self.extractfile(self._find_link_target(tarinfo))
  File "/usr/lib/python3.9/tarfile.py", line 2418, in _find_link_target
    raise KeyError("linkname %r not found" % linkname)
KeyError: "linkname 'micropython-umqtt.robust-1.0/umqtt/../../umqtt.simple/umqtt/simple.py' not found"
----------------------------------------
  WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /micropython-umqtt-robust/micropython-umqtt.robust-1.0.tar.gz/
[... 111 lines cut ...]
  File "/home/rob/apps/thonny/lib/python3.9/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 306, in _tweak_file
    with in_tar.extractfile(info) as f:
  File "/usr/lib/python3.9/tarfile.py", line 2120, in extractfile
    return self.extractfile(self._find_link_target(tarinfo))
  File "/usr/lib/python3.9/tarfile.py", line 2418, in _find_link_target
    raise KeyError("linkname %r not found" % linkname)
KeyError: "linkname 'micropython-umqtt.robust-1.0/umqtt/../../umqtt.simple/umqtt/simple.py' not found"
----------------------------------------
ERROR: Could not install packages due to an OSError: HTTPConnectionPool(host='127.0.0.1', port=36628): Max retries exceeded with url: /micropython-umqtt-robust/micropython-umqtt.robust-1.0.tar.gz/ (Caused by ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response')))

Error Command '['/home/rob/.cache/pipkin/workspaces/5178e1ea8c87b7e62566785d6a7356f2/bin/python3', '-I', '-m', 'pip', '--no-color', '--disable-pip-version-check', '--trusted-host', '127.0.0.1', 'install', '--no-compile', '--use-pep517', '--upgrade-strategy', 'only-if-needed', 'micropython-umqtt.robust', '--index-url', 'http://127.0.0.1:36628']' returned non-zero exit status 1.

```
The odd thing is that the package manager says version `1.0.1` is the latest stable version but the urls mentioned in the log refer to `1.0`

```cmd
mip.install('umqtt.robust')
```

does work

Any ideas where to start ?

Thanks
Rob

aivarannamaa, 2023-01-28:
Thank you for the report!

The tar.gz of `micropython-umqtt.robust` at PyPI seems to have unexpected structure, but given this file is outdated (https://pypi.org/project/micropython-umqtt.robust/#history), I'd rather not spend time on making Thonny understand, because with `mip` you're getting newer version.

After MicroPython 1.20 gets released, I'll refactor MicroPython package manager (pipkin) anyway and start using the new micropython-lib index (https://github.com/aivarannamaa/pipkin/issues/5).

WobblyRobbly, 2023-01-29:
Thanks for the update @aivarannamaa - That makes sense. Happy for this to closed then... look forward to the update

aivarannamaa, 2023-01-29:
Let's keep this open for now, then it reminds me to test the new solution with `umqtt.robust`

WobblyRobbly, 2023-01-29:
👍

aivarannamaa, 2023-08-10:
It looks like this was fixed in 4.1.1

closed 2023-08-10 by aivarannamaa