thonny/thonny #2630: Variables Dont working

Phenom4x, 2023-01-28:
The variables tab on thonny appear for some seconds when i click run code, and disappear.
dont show up even if the code stops or give an exeption.
Im using micropython for raspberry pi pico W
i tested on two different computers.

aivarannamaa, 2023-01-28:
Thank you for the report!

Are you using Thonny's simple mode (with large buttons and no menubar) or regular mode?

Which Thonny version? Which OS version?

Phenom4x, 2023-02-02:
> Thank you for the report!
> 
> Are you using Thonny's simple mode (with large buttons and no menubar) or regular mode?
> 
> Which Thonny version? Which OS version?

oh Sorry im having some poblems with my gmail
im using windows 10 64bit, i use the big buttons version, thonny 4.0.1, ui mode is extreme

aivarannamaa, 2023-02-02:
You are using the "Simple" UI mode and in this mode variables' pane is supposed to appear when you start debugging and stay hidden otherwise.

If you don't like this then I recommend switching to regular mode -- click on the link at upper-right corner of Thonny's window.

Phenom4x, 2023-02-02:
Ok, thanks

closed 2023-02-02 by aivarannamaa