thonny/thonny #2632: Back-end problem

elviine, 2023-01-29:
MacOS user here.

I decided to upgrade from Thonny 3.1.0 to Thonny 4.0.2 . I accidentally pressed decline when the program asked for my permission to access files and documents, and now it won't work at all and I don't know how to solve it.

PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_environment_info' (PermissionError: [Errno 1] Operation not permitted).
See Thonny's backend.log for more info.

Process ended with exit code 1.

How can I fix this?

Honpyt, 2023-01-31:
Same problem on MacOS 13.1 even when the installation process was completed as intended.

aivarannamaa, 2023-02-02:
Thank you for the report!

Please try adding "Full disk access" permission to Thonny:

* Go to System preferences => Security & privacy => Privacy
* Click on the padlock to allow changes
* Select "Full disk access"
* Click the "+" button and select Thonny (most likely under Applications).

aivarannamaa, 2023-02-12:
@elviine and @Honpyt, did my suggestion solve your issue?

elviine, 2023-02-12:
Yes it did! Thank you a lot.

closed 2023-02-12 by aivarannamaa