thonny/thonny #2633: Crash While changing the Interpreter

Vishnuxx, 2023-01-31:
Thonny is crashing on changing the interpreter type. 

Here is the full crash report:

{"app_name":"Python","timestamp":"2023-01-31 10:44:35.00 +0530","app_version":"3.10.9","slice_uuid":"e0b6e705-8ae0-304d-82c1-c8747742c3bf","build_version":"3.10.9","platform":1,"bundleID":"org.python.python","share_with_app_devs":0,"is_first_party":0,"bug_type":"309","os_version":"macOS 13.0 (22A380)","roots_installed":0,"name":"Python","incident_id":"63B1E007-2297-45F0-98C7-6574A3E06DDA"}
{
  "uptime" : 2200000,
  "procRole" : "Foreground",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "MacBookAir10,1",
  "coalitionID" : 225089,
  "osVersion" : {
    "train" : "macOS 13.0",
    "build" : "22A380",
    "releaseType" : "User"
  },
  "captureTime" : "2023-01-31 10:44:30.4508 +0530",
  "incident" : "63B1E007-2297-45F0-98C7-6574A3E06DDA",
  "pid" : 41455,
  "translated" : false,
  "cpuType" : "ARM-64",
  "roots_installed" : 0,
  "bug_type" : "309",
  "procLaunch" : "2023-01-31 10:44:19.2559 +0530",
  "procStartAbsTime" : 53940493575180,
  "procExitAbsTime" : 53940762225217,
  "procName" : "Python",
  "procPath" : "\/Users\/USER\/*\/Thonny.app\/Contents\/MacOS\/Python",
  "bundleInfo" : {"CFBundleShortVersionString":"3.10.9","CFBundleVersion":"3.10.9","CFBundleIdentifier":"org.python.python"},
  "storeInfo" : {"deviceIdentifierForVendor":"27388D2B-2D98-5BF1-80A0-7F0B43BC4BE0","thirdParty":true},
  "parentProc" : "launchd",
  "parentPid" : 1,
  "coalitionName" : "org.thonny.Thonny",
  "crashReporterKey" : "369147C6-B9AA-743F-8DF0-83B54966CBF0",
  "wakeTime" : 3625,
  "sleepWakeUUID" : "3FF637B6-F87A-4086-A820-657327BAE90A",
  "sip" : "enabled",
  "exception" : {"codes":"0x0000000000000000, 0x0000000000000000","rawCodes":[0,0],"type":"EXC_CRASH","signal":"SIGABRT"},
  "ktriageinfo" : "VM - pmap_enter retried due to resource shortage\nVM - pmap_enter retried due to resource shortage\nVM - pmap_enter retried due to resource shortage\n",
  "extMods" : {"caller":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"system":{"thread_create":4,"thread_set_state":0,"task_for_pid":5},"targeted":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"warnings":0},
  "faultingThread" : 0,
  "threads" : [{"triggered":true,"id":26471969,"threadState":{"x":[{"value":0},{"value":0},{"value":18446744073709551615},{"value":0},{"value":115},{"value":115},{"value":0},{"value":4388737276,"symbolLocation":0,"symbol":"Tcl_GetIndexFromObjStruct"},{"value":12100257483884249132},{"value":12100257488906954028},{"value":4386900312,"symbolLocation":0,"symbol":"PyModule_Type"},{"value":2},{"value":4374416096},{"value":4374416120},{"value":54},{"value":64},{"value":328},{"value":8078674376},{"value":0},{"value":6},{"value":7996564736,"symbolLocation":0,"symbol":"_main_thread"},{"value":259},{"value":7996564960,"symbolLocation":224,"symbol":"_main_thread"},{"value":1},{"value":4387010432,"symbolLocation":0,"symbol":"faulthandler_handlers"},{"value":0},{"value":0},{"value":105553169990912},{"value":7958540288,"symbolLocation":0,"symbol":"_OBJC_PROTOCOL_REFERENCE_$__NSActionBinderPlugin"}],"flavor":"ARM_THREAD_STATE64","lr":{"value":6462741740},"cpsr":{"value":1073745920},"fp":{"value":5637503712},"sp":{"value":5637503680},"esr":{"value":1442840704,"description":" Address size fault"},"pc":{"value":6462517796,"matchesCrashFrame":1},"far":{"value":4600070496}},"queue":"com.apple.main-thread","frames":[{"imageOffset":37412,"symbol":"__pthread_kill","symbolLocation":8,"imageIndex":0},{"imageOffset":27884,"symbol":"pthread_kill","symbolLocation":288,"imageIndex":1},{"imageOffset":268880,"symbol":"raise","symbolLocation":32,"imageIndex":2},{"imageOffset":2919768,"symbol":"faulthandler_fatal_error","symbolLocation":392,"imageIndex":3},{"imageOffset":17060,"symbol":"_sigtramp","symbolLocation":56,"imageIndex":4},{"imageOffset":204392,"symbol":"abort_with_payload_wrapper_internal","symbolLocation":104,"imageIndex":0},{"imageOffset":204288,"symbol":"abort_with_reason","symbolLocation":32,"imageIndex":0},{"imageOffset":218072,"symbol":"_objc_fatalv(unsigned long long, unsigned long long, char const*, char*)","symbolLocation":128,"imageIndex":5},{"imageOffset":217944,"symbol":"_objc_fatal(char const*, ...)","symbolLocation":44,"imageIndex":5},{"imageOffset":221336,"symbol":"AutoreleasePoolPage::badPop(void*)","symbolLocation":212,"imageIndex":5},{"imageOffset":8407352,"symbol":"-[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:]","symbolLocation":1636,"imageIndex":6},{"imageOffset":1385188,"symbol":"-[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:]","symbolLocation":556,"imageIndex":6},{"imageOffset":140576,"symbol":"CABackingStoreUpdate_","symbolLocation":224,"imageIndex":7},{"imageOffset":516732,"symbol":"invocation function for block in CA::Layer::display_()","symbolLocation":64,"imageIndex":7},{"imageOffset":136952,"symbol":"-[CALayer _display]","symbolLocation":1644,"imageIndex":7},{"imageOffset":133244,"symbol":"CA::Layer::display_if_needed(CA::Transaction*)","symbolLocation":788,"imageIndex":7},{"imageOffset":1588720,"symbol":"CA::Context::commit_transaction(CA::Transaction*, double, double*)","symbolLocation":468,"imageIndex":7},{"imageOffset":12264,"symbol":"CA::Transaction::commit()","symbolLocation":652,"imageIndex":7},{"imageOffset":3607584,"symbol":"-[_NSOrderOutAnimationProxyWindow initWithSnapshotOfWindow:]","symbolLocation":52,"imageIndex":6},{"imageOffset":2282524,"symbol":"-[_NSWindowTransformAnimation initWithWindow:type:interruptingAnimation:]","symbolLocation":520,"imageIndex":6},{"imageOffset":2281972,"symbol":"+[_NSWindowTransformAnimation windowTransformAnimationWithWindow:type:interruptingAnimation:]","symbolLocation":52,"imageIndex":6},{"imageOffset":8497328,"symbol":"-[NSWindow _doOrderWindow:]","symbolLocation":708,"imageIndex":6},{"imageOffset":814256,"symbol":"XUnmapWindow","symbolLocation":172,"imageIndex":8},{"imageOffset":828460,"symbol":"TkWmUnmapWindow","symbolLocation":76,"imageIndex":8},{"imageOffset":848156,"symbol":"TkpWmSetState","symbolLocation":268,"imageIndex":8},{"imageOffset":836708,"symbol":"Tk_WmObjCmd","symbolLocation":7000,"imageIndex":8},{"imageOffset":78504,"symbol":"TclNRRunCallbacks","symbolLocation":80,"imageIndex":9},{"imageOffset":82588,"symbol":"TclEvalEx","symbolLocation":1616,"imageIndex":9},{"imageOffset":40780,"symbol":"Tk_BindEvent","symbolLocation":4596,"imageIndex":8},{"imageOffset":63940,"symbol":"TkBindEventProc","symbolLocation":328,"imageIndex":8},{"imageOffset":94772,"symbol":"Tk_HandleEvent","symbolLocation":824,"imageIndex":8},{"imageOffset":96256,"symbol":"WindowEventProc","symbolLocation":112,"imageIndex":8},{"imageOffset":789928,"symbol":"Tcl_ServiceEvent","symbolLocation":136,"imageIndex":9},{"imageOffset":790624,"symbol":"Tcl_DoOneEvent","symbolLocation":284,"imageIndex":9},{"imageOffset":66420,"symbol":"Tk_TkwaitObjCmd","symbolLocation":424,"imageIndex":8},{"imageOffset":78504,"symbol":"TclNRRunCallbacks","symbolLocation":80,"imageIndex":9},{"imageOffset":19932,"symbol":"Tkapp_Call","symbolLocation":336,"imageIndex":10},{"imageOffset":590892,"symbol":"method_vectorcall_VARARGS","symbolLocation":124,"imageIndex":3},{"imageOffset":1798028,"symbol":"call_function","symbolLocation":132,"imageIndex":3},{"imageOffset":1761992,"symbol":"_PyEval_EvalFrameDefault","symbolLocation":19684,"imageIndex":3},{"imageOffset":1736024,"symbol":"_PyEval_Vector","symbolLocation":360,"imageIndex":3},{"imageOffset":1798028,"symbol":"call_function","symbolLocation":132,"imageIndex":3},{"imageOffset":1761992,"symbol":"_PyEval_EvalFrameDefault","symbolLocation":19684,"imageIndex":3},{"imageOffset":1736024,"symbol":"_PyEval_Vector","symbolLocation":360,"imageIndex":3},{"imageOffset":1798028,"symbol":"call_function","symbolLocation":132,"imageIndex":3},{"imageOffset":1766076,"symbol":"_PyEval_EvalFrameDefault","symbolLocation":23768,"imageIndex":3},{"imageOffset":1736024,"symbol":"_PyEval_Vector","symbolLocation":360,"imageIndex":3},{"imageOffset":543748,"symbol":"method_vectorcall","symbolLocation":124,"imageIndex":3},{"imageOffset":1798028,"symbol":"call_function","symbolLocation":132,"imageIndex":3},{"imageOffset":1766076,"symbol":"_PyEval_EvalFrameDefault","symbolLocation":23768,"imageIndex":3},{"imageOffset":1736024,"symbol":"_PyEval_Vector","symbolLocation":360,"imageIndex":3},{"imageOffset":1743780,"symbol":"_PyEval_EvalFrameDefault","symbolLocation":1472," [... cut ...]
Graphics: Apple M1, Apple M1, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: LPDDR4, Micron
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: Sep  3 2022 03:37:22 version 18.20.379.4.7.8.143 FWID 01-e3c71b50
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Bus: MacBook Air, Apple Inc.

philipheinser, 2023-02-01:
Same on Intel Mac with Ventura 13.2

Vishnuxx, 2023-02-05:
Hii , latest version of thonny has solved the issue.. try reinstalling Thonny

closed 2023-02-05 by Vishnuxx

chrkaatz, 2023-03-22:
I got the same error - tried 4.0.1 and 4.0.2 on MacOS Ventura 13.2.1?

Any recommendations?

anselanza, 2023-05-02:
Yup, happening for me too. Just downloaded Thonny today (first time installation) so I'm immediately stuck.