thonny/thonny #2634: 4.0.2 bash install on Raspberry Pi 4 Model B "failed to build cryptography"

SimUKdev, 2023-02-01:
I have been using Thonny 4.0.0 and 4.0.1 on a Raspberry Pi 4 Model B for a while, installed via the  `bash <(wget -O - https://thonny.org/installer-for-linux)` method as it's ahead of the Raspberry Pi OS version.

I wanted to update to 4.0.2 so repeated the same install procedure over the top of an existing install, and it failed with the "failed to build cryptography" and a lot of red text. To rule the "upgrading an existing install"  out of the reason for being the cause, and also to rule out the OS itself, I imaged an all new install of Raspberry Pi OS 11 Bullseye using the Raspberry Pi official imager.

On that new installation I did the usual `sudo apt get update` and `sudo apt get upgrade` to ensure it was fully up to date. I then did `sudo apt remove thonny` to remove the built-in version (which I believe is/was 4.0.1), after it was removed I did the `bash` install on that freshly imaged and updated Pi, but still got the same error: *(sorry for the long copy/paste of the terminal output)*

```
$ bash <(wget -O - https://thonny.org/installer-for-linux)

Redirecting output to ‘wget-log’.
Thonny only provides pre-built bundles for x86_64 (not armv7l).
I could install Thonny into a venv under your existing Python 3.9.2 (/usr/bin/python3).
Press ENTER to continue or Ctrl+C to cancel.
Creating virtual environment for Thonny
Marking virtual environment as private to Thonny
Installing Thonny's dependencies
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Ignoring adafruit-board-toolkit: markers 'sys_platform == "win32" or sys_platform == "darwin"' don't match your environment
Collecting jedi==0.18.*
  Downloading https://www.piwheels.org/simple/jedi/jedi-0.18.2-py2.py3-none-any.whl (1.6 MB)
     |████████████████████████████████| 1.6 MB 1.7 MB/s 
Collecting pyserial==3.5
  Downloading https://www.piwheels.org/simple/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
     |████████████████████████████████| 90 kB 1.6 MB/s 
Collecting pylint==2.15.*
  Downloading https://www.piwheels.org/simple/pylint/pylint-2.15.10-py3-none-any.whl (509 kB)
     |████████████████████████████████| 509 kB 2.7 MB/s 
Collecting docutils==0.19.*
  Downloading https://www.piwheels.org/simple/docutils/docutils-0.19-py3-none-any.whl (570 kB)
     |████████████████████████████████| 570 kB 3.3 MB/s 
Collecting mypy==0.991.*
  Downloading https://www.piwheels.org/simple/mypy/mypy-0.991-py3-none-any.whl (2.3 MB)
     |████████████████████████████████| 2.3 MB 3.1 MB/s 
Collecting asttokens==2.2.*
  Downloading https://www.piwheels.org/simple/asttokens/asttokens-2.2.1-py2.py3-none-any.whl (26 kB)
Collecting Send2Trash==1.8.1b0
  Downloading https://www.piwheels.org/simple/send2trash/Send2Trash-1.8.1b0-py3-none-any.whl (17 kB)
Collecting esptool==4.4.*
  Downloading https://www.piwheels.org/simple/esptool/esptool-4.4-py3-none-any.whl (288 kB)
     |████████████████████████████████| 288 kB 4.2 MB/s 
Collecting bcrypt==3.2.*
  Downloading https://www.piwheels.org/simple/bcrypt/bcrypt-3.2.2-cp39-cp39-linux_armv7l.whl (51 kB)
     |████████████████████████████████| 51 kB 154 kB/s 
Collecting cryptography==38.*
  Downloading cryptography-38.0.4.tar.gz (599 kB)
     |████████████████████████████████| 599 kB 1.0 MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting paramiko==2.12.*
  Downloading https://www.piwheels.org/simple/paramiko/paramiko-2.12.0-py2.py3-none-any.whl (213 kB)
     |████████████████████████████████| 213 kB 428 kB/s 
Collecting websockets==10.*
  Downloading https://www.piwheels.org/simple/websockets/websockets-10.4-cp39-cp39-linux_armv7l.whl (105 kB)
     |████████████████████████████████| 105 kB 1.1 MB/s 
Collecting ptyprocess==0.7.*
  Downloading https://www.piwheels.org/simple/ptyprocess/ptyprocess-0.7.0-py2.py3-none-any.whl (13 kB)
Collecting six
  Downloading https://www.piwheels.org/simple/six/six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting cffi>=1.1
  Using cached https://www.piwheels.org/simple/cffi/cffi-1.15.1-cp39-cp39-linux_armv7l.whl (364 kB)
Collecting ecdsa>=0.16.0
  Downloading https://www.piwheels.org/simple/ecdsa/ecdsa-0.18.0-py2.py3-none-any.whl (142 kB)
     |████████████████████████████████| 142 kB 399 kB/s 
Collecting reedsolo<=1.5.4,>=1.5.3
  Downloading https://www.piwheels.org/simple/reedsolo/reedsolo-1.5.4-cp39-cp39-linux_armv7l.whl (921 kB)
     |████████████████████████████████| 921 kB 1.2 MB/s 
Collecting bitstring<4,>=3.1.6
  Downloading https://www.piwheels.org/simple/bitstring/bitstring-3.1.9-py3-none-any.whl (39 kB)
Collecting parso<0.9.0,>=0.8.0
[... 231 lines cut ...]
  
  This package requires Rust >=1.48.0.
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
```

The text becomes red shortly after the (successful) pycparser stage, at the "ERROR:" line:

```
Building wheels for collected packages: cryptography
  Building wheel for cryptography (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
```

I have followed the advised `pip install --upgrade pip` and that updates pip ("*Successfully installed pip-23.0*"), but still get the same error when trying to install the Thonny 4.0.2 build as before.

```
(ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly)
```

I've not tried doing anything with Rust as also mentioned in the error message, and didn't have to do anything specific with Rust on a previous earlier Thonny 4.x install via the Bash, but it does seem from the error messages to be relating to something to do with (the lack of) that.

aivarannamaa, 2023-02-01:
Thank you for the report!

Thonny's installer scripts asks for `cryptography==38.*`, which means `pip` tries to install 38.0.4, but it seems that the build for this version has failed at [piwheels](https://www.piwheels.org/project/cryptography/).

Please try replacing `cryptography==38.*` with `cryptography==38.0.1` in the bash script and launching it again.

aivarannamaa, 2023-06-03:
Works with just released 4.1.0

closed 2023-06-03 by aivarannamaa