thonny/thonny #2638: code is not working in pico after running

Vishnuxx, 2023-02-05:
I have written a simple program to test whether builtin led in blinking or not.

```
   from machine import Pin
   led = Pin(25, Pin.OUT)
   import time

   led = machine.Pin(25, machine.Pin.OUT)

   while True:
       led.toggle()
       time.sleep(0.5)
```

#### To Reproduce

1. Install micropython in pico via thonny
2. Disconnect and reconnect the pico
3. Select the micropython interpreter for pico ( I've used both interpreters shown below)
[image]
4. Save to the board and click run

#### Output
Interprets without issues and the shell looks like
```
  >>> %Run -c $EDITOR_CONTENT
```
This is the expected behaviour But the **Led doesn't blink.**

Tested the same using arduino ide (c)  and vs code (micropython) and it worked

SimUKdev, 2023-02-06:
I just tried your code (had to remove indents from each line) on a Pico and it worked. On a Pico W it doesn't blink the LED, but changing the `25` to `"LED"` will allow the LED to blink on a Pico W.  Might be worth noting you have also defined the led twice in your code (line 2 and 5)

Do you have a Raspberry Pi "Pico" or a "Pico W"?

Vishnuxx, 2023-02-06:
yes, It was my mistake caused while pasting and formatting the code here. The code that I wrote in Thonny only define the Pin once. 
Thanks...You were right , I'm using Pico W.

closed 2023-02-06 by Vishnuxx