thonny/thonny #2639: Shift+Click selection starts from last mouse click instead of cursor position

rkagerer, 2023-02-05:
If I hold down SHIFT and then left-click somewhere in my code, my expectation is that it should highlight all of the text between the current cursor position and where I've clicked.

Instead, it highlights the text between where I've just clicked, and the last place I clicked before that.  i.e. It thinks the current cursor position is still located where I last dropped it using the mouse, when in fact I could have moved the cursor using the keyboard since that time.

To reproduce:

1) Paste the following code into a Thonny editor window (e.g. main.py):
```
a = 1
b = 2
c = a + b
d = a + b + c
```
2) Left-click on the third line, just after the "a"
3) Press HOME, or press the left arrow 5 times, to move the cursor to the beginning of that line.  (Don't use the mouse for this step!)
4) Hold down your left SHIFT button, and while holding it, click the end of the very last line.
5) Observe that even though you clearly intended to select the full contents of the final two lines, the highlight only grabbed part of the third line.

Repeat the above steps in any other text editor (e.g. Notepad, a Chrome TEXTAREA, etc) to see how it _should_ function.

Note this was tested using Thonny 4.0.2 on Windows 7.

This is my first Thonny bug report, please be gentle and forgive me if I didn't fill out the correct template or whatever.

aivarannamaa, 2023-02-11:
Thank you for this thorough report and sorry for the late reply!

I could reproduce this, but I'm afraid the problem is in the UI widget I've used, so it is not straightforward to fix. I'm considering moving to a new UI toolkit, so I'll postpone this issue for now.

mrexodia, 2023-05-02:
Investigating this a bit and I found something that could solve this while keeping tk: https://wiki.tcl-lang.org/page/Modern+Bindings+for+the+Text+Widget. Will investigate more and see if I can get this to work somehow.

mrexodia, 2023-05-02:
Managed to get it to work! Obviously this needs more testing to check for other side effects (maybe an experimental option?). Because I don't know Tcl I wanted to avoid rewriting `modernText`. I found if you use `tk.evalfile` during startup you can use `self.bindtags` to reuse the origin Tcl code.

mrexodia, 2023-05-02:
If someone wants to try this locally for the 4.0.2 release, look for `thonny.py` in your installation (on Windows this is in `%LocalAppData%\Programs\Thonny\Lib\site-packages\thonny`) and replace `workbench.py`, `tktexttext.py` and add `res\modernText.tcl` from the attached ZIP file.

[thonny-modernText-4.0.2.zip](https://github.com/thonny/thonny/files/11372836/thonny-modernText-4.0.2.zip)

rkagerer, 2023-05-02:
Thanks @mrexodia for taking a crack at this.  I tried it out, it loaded up under 4.0.2 on Windows OK.

It fixes the behavior of the selection when holding SHIFT, but breaks it when not holding SHIFT.  To reproduce use similar steps as above, but after step 3, try to highlight a new section of the document without holding down SHIFT and you'll see what I mean.  It's treating the old cursor point as the start of your selection.

mrexodia, 2023-05-02:
I see. Unfortunately I don’t know Tcl, but there might be an improved version of modernText out there that handles it (this version is from 2007). I also found that autocomplete wasn’t working (pressing Ctrl+Space), but I don’t know if it ever did.

aivarannamaa, 2025-02-04:
Need to investigate ntext:

* https://wiki.tcl-lang.org/page/ntext
* https://core.tcl-lang.org/tklib/doc/trunk/embedded/md/tklib/files/modules/ntext/ntext.md
* https://core.tcl-lang.org/tklib/doc/trunk/embedded/md/tklib/files/modules/ntext/ntextBindings.md
* https://core.tcl-lang.org/tklib/doc/trunk/embedded/md/tklib/files/modules/ntext/ntextIndent.md
* https://core.tcl-lang.org/tklib/doc/trunk/embedded/md/tklib/files/modules/ntext/ntextWordBreak.md
* https://github.com/tcltk/tklib/tree/master/modules/ntext

mrexodia, 2025-09-10:
@aivarannamaa will you merge #2768, but with `ntext` embedded? I do not really understand the logic behind leaving users with a bad experience for 2+ years after a fix has been proposed, but I am willing to give it a shot...