thonny/thonny #2647: Unable to install requests from manage packages section.

sounddrill31, 2023-02-10:
Device: ESP01S(flashed with the 2mb+ package) (Has the T25S80 chip, a quick google search points me to an 8mb spi flash.)
Firmware Version: esp8266-20220618-v1.19.1.bin(2mb+ version)

Thonny 8.6.12 Running on windows 10 with python 3.10.6

Error Message:

```Starting to apply changes to the target.
Copying requestsError (ManagementError(...), 'Command output was not empty')```

sounddrill31, 2023-02-10:
```Looking in indexes: http://127.0.0.1:36628
127.0.0.1 - - [10/Feb/2023 13:59:01] "GET /requests/ HTTP/1.1" 200 -
Collecting requests
  Using cached http://127.0.0.1:36628/requests/requests-2.28.2-py3-none-any.whl/
127.0.0.1 - - [10/Feb/2023 13:59:02] "GET /charset-normalizer/ HTTP/1.1" 200 -
Collecting charset-normalizer<4,>=2
  Using cached http://127.0.0.1:36628/charset-normalizer/charset_normalizer-3.0.1-cp310-cp310-win_amd64.whl/
127.0.0.1 - - [10/Feb/2023 13:59:02] "GET /certifi/ HTTP/1.1" 200 -
Collecting certifi>=2017.4.17
  Using cached http://127.0.0.1:36628/certifi/certifi-2022.12.7-py3-none-any.whl/
127.0.0.1 - - [10/Feb/2023 13:59:03] "GET /idna/ HTTP/1.1" 200 -
Collecting idna<4,>=2.5
  Using cached http://127.0.0.1:36628/idna/idna-3.4-py3-none-any.whl/
127.0.0.1 - - [10/Feb/2023 13:59:03] "GET /urllib3/ HTTP/1.1" 200 -
Collecting urllib3<1.27,>=1.21.1
  Using cached http://127.0.0.1:36628/urllib3/urllib3-1.26.14-py2.py3-none-any.whl/
Installing collected packages: charset-normalizer, urllib3, idna, certifi, requests
Successfully installed certifi-2022.12.7 charset-normalizer-3.0.1 idna-3.4 requests-2.28.2 urllib3-1.26.14
Starting to apply changes to the target.
Copying certifiError (ManagementError(...), 'Command output was not empty')

sounddrill31, 2023-02-10:
New error after flashing 1MB firmware:

```127.0.0.1 - - [10/Feb/2023 14:27:33] "GET /idna/ HTTP/1.1" 200 -
Collecting idna<4,>=2.5
  Using cached http://127.0.0.1:36628/idna/idna-3.4-py3-none-any.whl/
Installing collected packages: charset-normalizer, urllib3, idna, certifi, requests
Successfully installed certifi-2022.12.7 charset-normalizer-3.0.1 idna-3.4 requests-2.28.2 urllib3-1.26.14
Starting to apply changes to the target.
Copying charset_normalizer................
Copying certifi...Error Could not write next block after having written 5120 bytes to /lib/certifi/cacert.pem

aivarannamaa, 2023-02-10:
This package is not meant for MicroPython

closed 2023-02-10 by aivarannamaa

sounddrill31, 2023-02-11:
> This package is not meant for MicroPython

Thanks for the response, much appreciated!

sounddrill31, 2023-02-11:
Loaded micropython-urequests and rewrote code, and now project works!