thonny/thonny #2648: Issue with plugins

mrtechnorea, 2023-02-10:
Hello,

I have downloaded Thonny.
I wanted to install numpy or matplotlib plugins.
I used the manage plugin through the Tools menu.
I upgraded the dependencies.
The plugin installs but I always get the same error :
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'numpy'

Whether I install it for me or for all users. 

I didn't change anything during the installation or after. Just the download of numpy.

I test it by writing:

import numpy as np
and nothing more.

I am having this problem and have tried multiple times on different computers and even with the portable version.

# Frontend plugin is in cpython_frontend.py
import ast
import builtins
import functools
import importlib
import inspect
import io
import os.path
import queue
import re
import site
import subprocess
import sys
import tokenize
import traceback
import types
import warnings
from typing import Dict, List, Optional, Tuple, Union

import __main__
import thonny
from thonny import report_time
from thonny.backend import MainBackend, logger
from thonny.common import (
    OBJECT_LINK_END,
    OBJECT_LINK_START,
    REPL_PSEUDO_FILENAME,
    STRING_PSEUDO_FILENAME,
    BackendEvent,
    CommandToBackend,
    DistInfo,
    EOFCommand,
    FrameInfo,
    ImmediateCommand,
    InlineCommand,
    InlineResponse,
    InputSubmission,
    MessageFromBackend,
    TextRange,
    ToplevelCommand,
    ToplevelResponse,
    UserError,
    ValueInfo,
    execute_system_command,
    execute_with_frontend_sys_path,
    get_augmented_system_path,
    get_exe_dirs,
[... 1288 lines cut ...]
                line = "\n".join(partlines) + "\n"

            yield (line, None, None, None)

    items = rec_format_exception_with_frame_info(e_type, e_value, e_traceback)

    return list(items)

def in_debug_mode():
    return os.environ.get("THONNY_DEBUG", False) in [1, "1", True, "True", "true"]

def _is_library_file(filename):
    return (
        filename is None
        or path_startswith(filename, sys.prefix)
        or hasattr(sys, "base_prefix")
        and path_startswith(filename, sys.base_prefix)
        or hasattr(sys, "real_prefix")
        and path_startswith(filename, getattr(sys, "real_prefix"))
        or site.ENABLE_USER_SITE
        and path_startswith(filename, site.getusersitepackages())
    )

def get_backend():
    return _backend

aivarannamaa, 2023-02-10:
"Tools => Manage plug-ins" is meant for installing packages which add extra feature for Thonny itself (as explained by the text at the top of the dialog).

If you want to install extra packages to be used in your own programs, then you need to use "Tools => Manage packages".

closed 2023-02-10 by aivarannamaa

mrtechnorea, 2023-02-10:
thank you

-- 

*REA Gilles*

*Référent numérique *

Digital adviser

 ***@***.***

*Lycée Français René Descartes*

96, Rue Christopher Howes, Wat Phnom BP 1132

Phnom Penh Royaume du Cambodge
<https://www.facebook.com/lfrdpp>  <https://www.instagram.com/lfrdpp>
<https://www.linkedin.com/school/lfrd/>