thonny/thonny #2649: Could not find volume CIRCUITPY

georgikoemdzhiev, 2023-02-10:
I am unable to write to the device using the IDE. I am getting the below error when I press the run button:

```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'write_file' (RuntimeError: Could not find volume CIRCUITPY).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your CircuitPython device and try again.

Process ended with exit code 1.
```

Version of Thonny: 4.0.2
Version of CircuitPython: 8.0.0

aivarannamaa, 2023-02-10:
Thanks for the report!

Do you have a removable drive named "CIRCUITPY" in your system? Which OS are you using?

georgikoemdzhiev, 2023-02-10:
> Thanks for the report!
> 
> Do you have a removable drive named "CIRCUITPY" in your system? Which OS are you using?

No problem. 
Yes, I do have a removable drive "CIRCUITPY" on my system when the error is generated in Thnny.
I am using EndeavourOS Linux x86_64 (Arch Linux)

aivarannamaa, 2023-02-11:
What is the absolute path of this directory?

georgikoemdzhiev, 2023-02-11:
> What is the absolute path of this directory?

It is `/run/media/$USER/CIRCUITPY`

aivarannamaa, 2023-02-11:
Please post what do you get when you execute command `mount` in terminal, while your CircuitPython device is plugged in.

georgikoemdzhiev, 2023-02-11:
> 

Here is the output of the command:

```log
[georgi@xps159560 ~]$ mount
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
sys on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
dev on /dev type devtmpfs (rw,nosuid,relatime,size=8104192k,nr_inodes=2026048,mode=755,inode64)
run on /run type tmpfs (rw,nosuid,nodev,relatime,mode=755,inode64)
efivarfs on /sys/firmware/efi/efivars type efivarfs (rw,nosuid,nodev,noexec,relatime)
/dev/nvme0n1p2 on / type btrfs (rw,noatime,compress=zstd:3,ssd,space_cache=v2,subvolid=256,subvol=/@)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev,inode64)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
bpf on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=34,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=17942)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime,pagesize=2M)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
ramfs on /run/credentials/systemd-sysusers.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
ramfs on /run/credentials/systemd-tmpfiles-setup-dev.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
/dev/nvme0n1p2 on /home type btrfs (rw,noatime,compress=zstd:3,ssd,space_cache=v2,subvolid=257,subvol=/@home)
/dev/nvme0n1p2 on /var/cache type btrfs (rw,noatime,compress=zstd:3,ssd,space_cache=v2,subvolid=258,subvol=/@cache)
/dev/nvme0n1p2 on /var/log type btrfs (rw,noatime,compress=zstd:3,ssd,space_cache=v2,subvolid=259,subvol=/@log)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,nr_inodes=1048576,inode64)
/dev/nvme0n1p1 on /boot/efi type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)
ramfs on /run/credentials/systemd-tmpfiles-setup.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
ramfs on /run/credentials/systemd-sysctl.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=1623076k,nr_inodes=405769,mode=700,uid=1000,gid=1000,inode64)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
```

aivarannamaa, 2023-02-11:
Thanks!

If you say that the device's filesystem was present at `/run/media/$USER/CIRCUITPY` while you got this output from `mount`, then I'm out of ideas, as I expected to see this path in this output as well.

I guess I need to find another way for determining mounted volumes. Does any of the options described at https://linuxopsys.com/topics/list-mounted-drives-on-linux give `/run/media/$USER/CIRCUITPY` ?

georgikoemdzhiev, 2023-02-12:
I think it may have been entirely my fault. Today I noticed that when I launch the editor as a regular user (i.e. not `root`) it gave me a permission denied Error 13 and suggested I add my user to the `uucp` group. After that, I was able to run my script fine! 

Thank you for your time looking into this, I apologise that it took that long for me to notice the error in the IDE.

aivarannamaa, 2023-02-12:
So you got this error when running Thonny `root`, right?

georgikoemdzhiev, 2023-02-12:
I thought the problem went away. However, when I disconnected the CIRCUIT python device (Raspberry pi Pico in my case), the issue started occurring again. 

That is an as normal user with the user being part of `uucp` group as the IDE suggested.  

Perhaps, during the process of disconnecting the device, the IDE cannot detect it anymore

aivarannamaa, 2023-02-12:
The problem occurs because Thonny uses the output of `mount` command for detecting the mounted volumes, but in your case `CIRCUITPY` is missing in this output.

If you can confirm that another approach for detecting the volumes works in your system (eg. something from https://linuxopsys.com/topics/list-mounted-drives-on-linux), then I would consider changing Thonny's approach.

georgikoemdzhiev, 2023-02-17:
Here is what I discovered:

1. I started Thonny 
2. Plugged my Curquit Python device into my PC
3. Pressed the PAUSE button in Thonny to restart the backend (i.e. so that my device is registered in Thonny)
4. I opened `code.py` file from my Curquit Python device and did some changes
5. Pressed Save - > that resulted in the originally reported error
6. I opened a file explorer to check if the `CIRCUITPY` directory was there, it was 
7. I opened a terminal and executed `mount | grep CIRCU` but nothing came out (i.e. no match/no directory mounted)
8. I went back to Thonny and tried to save the changes in `code.py` but I got the same error
9. I went to the File explorer and the `CIRCUITPY` directory was still there, I clicked it and the path that was shown was `/run/media/$USER/georgi/CIRCUITPY`
10. I went back to Thonny but tried to save the changes and this time it worked!

Observations, so it seems that the removable  `CIRCUITPY` drive is not actually mounted until I interact with it from the file explorer (i.e. when I open it).

I am not sure if that helps or not. 

P.S.: apologies for the slow response.

jwillikers, 2023-02-26:
The `uucp` group is meant for `tty` i.e. serial port access and not for mounting filesystems. UDisks2 is typically used for mounting filesystems in userspace and preferring it over `mount` would likely solve any related permissions errors. Related issues: flathub/org.thonny.Thonny/#153 and #2650

PR #2683 (2023-03-26): Use UDisks to find mounted filesystems on Linux
This should generally work better on Linux than using the mount command directly. UDisks is the utility that all major Linux desktop distributions use to handle mounting filesystems in userspace. UDisks is accessible through a D-Bus API on the system bus.

D-Bus is complex and requires adding a dependency on a Python library for handling asynchronous communication with the bus. I chose dbus-next since it is actively developed, modern, and doesn't require a glib main loop. Since Thonny is not a Gtk application, it makes more sense to use native Py [... cut]
closed 2023-03-26 by aivarannamaa