thonny/thonny #2650: Thonny on raspberrry pi running pop_os gives error when looking for connected pico

stormagedon92, 2023-02-10:
when i click the "Local Python 3 - /usr/bin/python3" i get this error message:

Traceback (most recent call last):
  File "/app/lib/python3.10/site-packages/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/app/lib/python3.10/site-packages/thonny/workbench.py", line 865, in _post_backend_menu
    if uf2_device_is_present_in_bootloader_mode():
  File "/app/lib/python3.10/site-packages/thonny/plugins/micropython/uf2dialog.py", line 602, in uf2_device_is_present_in_bootloader_mode
    for vol in list_volumes(skip_letters=["A"]):
  File "/app/lib/python3.10/site-packages/thonny/misc_utils.py", line 101, in list_volumes
    mount_output = subprocess.check_output("mount").splitlines()
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'mount'

aivarannamaa, 2023-02-11:
Thank you for the report!
What do you get if you run command 'mount' in terminal?

stormagedon92, 2023-02-11:
Thanks for the quick reply, this is what I get:

sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,nosuid,relatime,size=3788624k,nr_inodes=947156,mode=755,inode64)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,noexec,relatime,size=794432k,mode=755,inode64)
/dev/mmcblk0p2 on / type ext4 (rw,relatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev,inode64)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k,inode64)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
bpf on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=29,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=18136)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime,pagesize=2M)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
none on /run/credentials/systemd-sysusers.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
/var/lib/snapd/snaps/bare_5.snap on /snap/bare/5 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide)
/var/lib/snapd/snaps/core_14452.snap on /snap/core/14452 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide)
/var/lib/snapd/snaps/core18_2681.snap on /snap/core18/2681 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide)
/var/lib/snapd/snaps/core20_1782.snap on /snap/core20/1782 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide)
/var/lib/snapd/snaps/core20_1826.snap on /snap/core20/1826 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide)
/var/lib/snapd/snaps/firefox_2310.snap on /snap/firefox/2310 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide)
/var/lib/snapd/snaps/gnome-3-38-2004_122.snap on /snap/gnome-3-38-2004/122 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide)
/var/lib/snapd/snaps/gtk-common-themes_1535.snap on /snap/gtk-common-themes/1535 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide)
/var/lib/snapd/snaps/snap-store_639.snap on /snap/snap-store/639 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide)
/var/lib/snapd/snaps/snapd_17954.snap on /snap/snapd/17954 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide)
/dev/mmcblk0p2 on /var/snap/firefox/common/host-hunspell type ext4 (ro,noexec,noatime)
/var/lib/snapd/snaps/tiled_3948.snap on /snap/tiled/3948 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide)
/dev/mmcblk0p1 on /boot/firmware type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,nosuid,nodev,noexec,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=794428k,nr_inodes=198607,mode=700,uid=1000,gid=1000,inode64)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
portal on /run/user/1000/doc type fuse.portal (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
tmpfs on /run/snapd/ns type tmpfs (rw,nosuid,nodev,noexec,relatime,size=794432k,mode=755,inode64)
nsfs on /run/snapd/ns/firefox.mnt type nsfs (rw)

aivarannamaa, 2023-02-11:
Strange... for some reason you can execute `mount` in terminal without problems (you didn't use sudo, right?), but when Thonny tries to execute it (` subprocess.check_output("mount")`), this program is not found (`No such file or directory: 'mount'`).

Please send me the outputs of two more commands:

* `echo $PATH`
* `which mount`

jwillikers, 2023-02-26:
@aivarannamaa This is an issue with the Flatpak. There is no `mount` executable inside the Flatpak due to the sandbox. I document further details of why in flathub/org.thonny.Thonny#153. TLDR Thonny should prefer mounting devices on Linux using the userspace mounting utility UDisks2 via its [DBUS API](http://storaged.org/doc/udisks2-api/latest/gdbus-org.freedesktop.UDisks2.Filesystem.html). I'm looking into the best way to accomplish this right now. It looks to me like using [python-dbus-next](https://github.com/altdesktop/python-dbus-next) for making DBUS calls would be best. All the other Python DBUS libraries appear dead and/or have significant dependencies on Gtk-based components.