thonny/thonny #2651: invalid header: 0xffffffff

Anton-Wez, 2023-02-11:
Dicsription 1

#Error Message using Thonny.IDE on rasphberry Pi after flashing with GENERIC_S3-20220618-v1.19.1.bin from Micropython.org

Author: Anton Herbertshagen
Date: 10_JAN-2023

Model: Wifi_Kit_32_V3
Chip: ESP32-S3

----------------------------------------------------------------------------------------------------------

invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff

invalid header ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049b21

invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff

onnection lost -- device reports readiness to read but returned no data (device disconnected or multiple access on port?)

Process ended with exit code 1.

aivarannamaa, 2023-02-11:
Can you get the REPL on this device via a simpler tool (Putty, tio, TeraTerm, screen...)? If not then make sure you have installed correct MicroPython variant with correct [flash mode](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/spi-flash-modes.html#spi-flash-modes).

Anton-Wez, 2023-02-11:
I´m not sure how i get it done..can you guide me through?

aivarannamaa, 2023-02-11:
> I´m not sure how i get it done (REPL)..can you guide me through?

Nevermind, I think this is not so important.

> I´ve tried to flash with all of the Micropython bin data regarding ESP32-S3 using thonny.ide with the same error result

Did you try different flash modes?