thonny/thonny #2654: Thonny files open in multiple windows - related to issue #2176

jrs69, 2023-02-12:
I have Thonny 4.0.1.
Allow only single Thonny instance is checked.

If I open multiple files from the CLI or file manger, files are opened in Tabs - all good.

If I open multiple files in a Linux script For loop, they open in multiple windows, and I'm assuming, multiple instances  - this will no doubt be obvious to a Linux expert -  I have no clue!

aivarannamaa, 2023-02-12:
Please try whether the issue still exists when you add a small delay (e.g. 1 sec) in your for loop to allow Thonny to start up properly.

But if you just want to open several files at once, you can add them as separate parameters to the same `thonny` command.

jrs69, 2023-02-12:
Thks, that fixed the problem, script below.

===============================================================

ext=".py"

if [ -z "$1" ] #$1 is null
then
  /home/june/apps/thonny/bin/thonny &>/dev/null & #willl open last opened file
else
  for fname in "$@"
      do
      if [[ ! "$fname" == *"."* ]]; # fname doesn't contain dot     
      then
         fname+="$ext"
      fi
      touch $fname
      chmod u+x $fname
      sleep 1
      /home/june/apps/thonny/bin/thonny $fname  &>/dev/null &
   done
fi

aivarannamaa, 2023-02-12:
Did you trying sending all parameters to `thonny` at once, without the loop?

jrs69, 2023-02-12:
Yes, that works correctly, the script:
Handles the absence/presence of .py extension
Sets permissions to run the prog from the CLI
Creates a new named file

closed 2023-02-12 by aivarannamaa