thonny/thonny #2656: Does not maximize after minimization on Mac devices.

Ivinmathew1, 2023-02-13:


aivarannamaa, 2023-02-15:
Do you mean that you can't get Thonny back by clicking on Thonny's icon in the dock? 

I'm not a Mac user myself so I didn't even know that it should work like this, but it does work like this with Chrome.

I'm afraid this is a shortcoming of Thonny's UI toolkit (Tk) and you need to maximize it by clicking on the small preview icon on the dock, which represens Thonny's main window. Does this work for you?

Ivinmathew1, 2023-02-15:
Yes, this is exactly what the problem is. I don't have that preview section on my Mac so I do not have a way to maximize it.

aivarannamaa, 2023-02-15:
I could reproduce this by checking "Minimise windows into application icon" in Dock settings.

The best work-around I could find, was to right-click the Thonny icon in the Dock, select "Show all windows" and then click on the white rectangle representing Thonny's window.

Ivinmathew1, 2023-02-15:
This works!