thonny/thonny #2657: Using the site-package’s version of ESPtool from within Thonny?

brianlinuxing, 2023-02-13:
Am using Thonny on a Ubuntu 18.04 based system.

Thonny has been successfully installed with the bash script and everything works well.

I noticed this directory: /home/brian/apps/thonny/lib/python3.10/site-packages/esptool

**Is there a clever trick or method that would allow me to invoke the site-package’s version of ESPtool from within Thonny?**

I did study the documentation but couldn’t see anything obvious.

[Context: I’m trying to debug an issue with an ESP32S3 module.]

Thank you :)

aivarannamaa, 2023-02-15:
Do you mean getting easy access to this tool via command line? You could select "Thonny's Python" as Thonny's interpreter from the lower-right corner of Thonny's main window, then select "Tools => Open system shell" and then launch esptool with `python -m esptool`

brianlinuxing, 2023-02-15:
Thank you @aivarannamaa

The reason I ask is, I have several ESP32 devices (under Linux), once simply plugged in they work perfectly in Thonny, talks well to ttyACM0.

Yet if I exit thonny and run esptool (installed by PIP or standalone) it timeouts with "No serial data received."

I assume that either:

1. Thonny has a more robust method of communicating with the ESP32 devices?

or 2. That they (Thonny and ESPtool) are, essentially, talking to different parts (and in different ways) of the ESP32 and number 1 doesn't apply?

**I did briefly scan the thonny ESP backend code, but I would welcome your insights on how best to talk to ESP32 devices.**

[Yes, I know I *could* fiddle with the boot/reset and put it manually into Bootloader Mode, but I much prefer to do it programmatically and I might not always have physical access to the buttons.]

aivarannamaa, 2023-02-15:
That's how Thonny composes the `esptool` command line: https://github.com/thonny/thonny/blob/4f422f0281c8d3eb944b417e8a370a00e5d1d397/thonny/plugins/esp/__init__.py#L288

In case Thonny has been connected to the same port before starting the installation procedure, it closes this connection first.

aivarannamaa, 2023-02-15:
> Yes, I know I could fiddle with the boot/reset and put it manually into Bootloader Mode

I've never worried about the bootloader mode with ESP -- `esptool` has managed it transparently without problems for me.

brianlinuxing, 2023-02-15:
It does seem to be a wider issue with ESP32s (I've spend hours noting it across various forums) and people even put oscilloscopes on their devices.

I had assumed it was a cabling issue but after the 5th one, now I am less convinced :)

I am trying to reduce the variables and ironically it is not flashing that is the issue, done 3+ devices without any problems

It is just doing a status check (here in Windows, working well with thonny!)

"esptool --port COM4 flash_id
esptool.py v4.5
Serial port COM4
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html"

[Naturally I have tried various permutations, baud rate, etc]

Same ESP32 device on a Pi 400 running Diet Pi (thonny is great on it!)

"esptool.py -b 115200 --port /dev/ttyACM0  flash_id
esptool.py v4.5
Serial port /dev/ttyACM0
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html"

Be nice if Thonny could invoke esptool in a none destructive fashion  to report on the current device, basically chip_id

**Would that be possible?**

I am very open to suggestions!

aivarannamaa, 2023-02-16:
I think it wouldn't be too hard to add a "Get device info" button to this dialog.

aivarannamaa, 2023-03-26:
Handled under #2707. The esptool GUI in Thonny 4.1 will have "Query device info" command next to "Install" button.

closed 2023-03-26 by aivarannamaa