thonny/thonny #2658: Plugin for Classroom Use

graham-mitchell-vcs, 2023-02-13:
@aivarannamaa would you be interested in paid development of a plugin to enhance collaborative use in classrooms?

We have a decent budget (five figures) and would like the plugin to be publicly available for free afterwards. We can share wireframes and workflow via a Zoom call.

Just a reminder, we are the team @ Core Electronics.

aivarannamaa, 2023-02-15:
Thanks for the proposal!

I'm not sure I can find time for this development, but I'd be interested to hear more. Lets arrange the call via e-mail. My e-mail address is aivar.annamaa@gmail.com

graham-mitchell-vcs, 2023-02-15:
Thanks Aivar, I've sent you an email and will mark this issue as solved so it doesn't distract others :)

closed 2023-02-15 by graham-mitchell-vcs

aivarannamaa, 2023-02-15:
@graham-mitchell-vcs, I still haven't received your e-mail, neither in my inbox nor in the spam folder.

graham-mitchell-vcs, 2023-02-15:
Interesting! Please send an email direct to support@coreelectronics.com.au if you can't see this one that was sent earlier today:

[image]