thonny/thonny #2659: Cannot write to SD card when SD card is open in thonny's file system

JuanDelaCruz24k, 2023-02-14:
I am using CircuitPython 7.3.3 / 8.0.0 on a RP2040 board. After mounting the SD card, if I don't chdir to the sd card, writing will push thru  but will fail once I chdir to the SD card and open in thonny's file system.

`>>> %Run -c $EDITOR_CONTENT
    Traceback (most recent call last):
      File "<stdin>", line 4, in <module>
    OSError: [Errno 30] Read-only filesystem
>>>
`
[image]

aivarannamaa, 2023-02-14:
By default CircuitPython only allows writing via the mounted volume and keeps the filesystem read-only for MicroPython code, regardless of whether it is being used via Thonny or something else.

JuanDelaCruz24k, 2023-02-15:
Maybe I am not clear. Writing to the SD card via python code fails when thonny had refresh it's directory listing and is already showing on the list of mounted volume in the device (left bar @ bottom) see first screenshot above. Writing to the SD card via code DOES NOT fail when  thonny had not access the mounted volume to display on the left bar. The screenshot below shows code running and writing on the SD card.

[image]

aivarannamaa, 2023-02-15:
Oh, sorry, I was thinking about the built-in flash of the CircuitPython device and somehow didn't notice you were talking about the external flash card.

I can't think how listing the files can affect this. That's what Thonny's file browser executes on the device: https://github.com/thonny/thonny/blob/4f422f0281c8d3eb944b417e8a370a00e5d1d397/thonny/plugins/micropython/mp_back.py#L1086

How do you mount the SD card? Can you share the code? Which device are you using? Perhaps it's a CircuitPython's bug.

If you are able to reproduce this problem (i.e. locking the SD just by doing `os.listdir("/")` and `os.stat("/sd")`)  outside of Thonny (e.g. via `tio`), then I think it's worth raising an issue in the CircuitPython tracker.

JuanDelaCruz24k, 2023-02-16:
Thanks for your reply. 

I have tried using VS Code and writing to the SD card via python code and it works just fine even when the folder CIRCUITPY is open. I also tried using macos|screen|REPL and it also works just fine. see screen shot below. I have also attached the code that I am using.

[image]

mountSDcard

import sys
import os

import adafruit_sdcard
import board
import busio
import digitalio
import storage

spi = busio.SPI(board.GP18, board.GP19, board.GP16)
cs = digitalio.DigitalInOut(board.GP17)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

sys.path.append("/sd")
sys.path.append("/sd/lib")

print("SD card mounted.")

=== logTemp ===

import microcontroller
from time import sleep

with open("/sd/YourText.txt", "w") as writefile:
    print("The first line of text on SD Card", file=writefile)
    writefile.close()

with open("/sd/YourText.txt", "a") as appendfile:
    print("Check temperature data on SD Card", file=appendfile)
    appendfile.close

with open("/sd/YourText.txt", "r") as inputfile:
    for line in inputfile:
        print(line)
    inputfile.close

while True:
    with open("/sd/Temp_Data_Logging.txt", "a") as appendfile:
        print(microcontroller.cpu.temperature, file=appendfile)
    appendfile.close
    sleep(5)

aivarannamaa, 2023-02-16:
Please also check whether `os.stat("/sd")` somehow affects the outcome.

JuanDelaCruz24k, 2023-02-17:
So far, every test and commands that you have suggested only fails inside of Thonny and works just fine using other editors. If you are interested in lookin at the problem then the community will thank you. I don't personally use Thonny but my kid was having this problem. He had come to terms and said he is already using another editor. We thank you for your attention.

[image]

aivarannamaa, 2023-02-17:
Which RP2040 board is this?

Does the problem persist when you close Thonny's file browser and restart the device?