thonny/thonny #2660: Installing packages via the Manage Packages dialog doesn't seem to work.

allstar87, 2023-02-15:
Running the flatpak version of Thonny (4.0.2) on Linux Mint 21.1 Cinnamon Edition.

Working with the PyGObject module, first line of my script is "import gi" I get a ModuleNotFoundError: No module named "gi"

The Manage Packages system doesn't seem to work, searching for and installing a module that way...nothing seems to happen.  The module name doesn't appear in the list on the left, and I still get that error running my script.

Works fine on the very obsolete .deb package (3.3.14)

aivarannamaa, 2023-02-15:
> The Manage Packages system doesn't seem to work, searching for and installing a module that way...nothing seems to happen. The module name doesn't appear in the list on the left, and I still get that error running my script.

What's the exact name of the package you were trying to install?

allstar87, 2023-02-15:
PyGObject.  As seen in the attached screenshot.  Though I just tried pygame as well and that also didn't work, it doesn't appear in the list to the right and if I try to import it I get that ModuleNotFoundError.
[image]

aivarannamaa, 2023-02-16:
The screenshot shows the search results for `pygobject`. I didn't understand what happens when you open the first link and click "Install"?

allstar87, 2023-02-16:
Click the first link, the one that says PyGObject, that's rendered in bold there.  Click Install.  A small window pops up, it flashes text that's too fast to read, closes, and nothing seems to have gotten done.

aivarannamaa, 2023-02-16:
> A small window pops up, it flashes text that's too fast to read, closes, and nothing seems to have gotten done.

I see, now it gets interesting. This small window should have an underlined link near the bottom edge, left from the buttons. If you are able to click the link before the window closes, then you can read and copy the text.