thonny/thonny #2667: RP2040/Pico shell: integers starting with 0 don't print

linalg137, 2023-02-21:
When I enter any integer starting with zero into Thonny's MicroPython REPL, it doesn't print the value:

[image]

However, it really does have a value:
```
>>> print(04)
4
>>> 
```
Also, it works in other programs:

[image]

This might have something to do with octal.

aivarannamaa, 2023-03-07:
Thanks for the report!

This happens because Thonny's REPL doesn't use MicroPython REPL directly and therefore needs to use its own tricks for printing the value of the expression when entered into shell. In this process it uses CPython's parser and this fails as CPython doesn't allow leading zeroes like this (you need to write `0o4` instead of `04` to express octal).

I'm surprised that MicroPython and CPython differ in this respect and I recommend simply to avoid leading zeroes like this, because it would cause confusion.

aivarannamaa, 2023-03-07:
I created a related discussion here: https://github.com/micropython/micropython/discussions/10967

closed 2023-03-11 by aivarannamaa