thonny/thonny #2670: Autocomplete issue

slavomirMatuska, 2023-02-24:
I have problem with the Auto-Completion (AC). I`m using Thoony for programming the NodeMCU board with "MicroPython v1.19.1 on 2022-06-18; ESP module with ESP8266". I know that AC is invoked only when ctrl+space is pressed. But AC works for example on the package time, but when i try AC on uos, it is not working. I also tried other modules from the micropython, but without luck. In the Object inspector i can see all function of the uos package, but AC still not working (see picture). I tried install Thonny on the Linux and Windows machine as well and it did not work. For Linux i also tried install Thonny from script or as a python package. What am I missing?
Thanks.
[image]

aivarannamaa, 2023-03-11:
Sorry for the late reply!

I could not reproduce your issue with ESP32 on Windows machine. Will try Linux later.

You could check, whether something interesting shows up in frontend.log or backend.log (Tools => Open Thonny data folder).

mdrescher, 2023-03-14:
I can confirm the issue, using Thonny 4.0.2, ESP8266 DevKitC running Micropython 1.19.1 using MacBook Air with MacOS Ventura.

The editor and shell only autocomplete (AC) standard CPython tokens, such as while, import, True, False, etc. No ESP8266 tokens such as machine, uos, etc. are autocompleted. I have **NOT** tested this with an ESP32 (they are packed away somewhere).

However, when opening a System Shell (Tools --> Open System Shell) I can press Tab to do autocomplete on all tested CPython and MicroPython language tokens. It ACs import, uos, and many other tokens I tested: 
[image]

It would be great if you could integrate @josverl's [micropython-stubs](https://github.com/Josverl/micropython-stubs) as that would instantaneously give you (well, jedi) access to a huge host of MicroPython versions on a huge selection of boards (I think you are already looking into that)

slavomirMatuska, 2023-03-27:
Thanks for the reply, 
i`m using Thonny for teaching students. Non of the students were able to make the autocomplete work on their own laptops (with ESP8266). Maybe we are missing something. Could you please wrote us step-by-step process how to install Thonny and how to enable autocomplete when working with ESP8266? 
Thakns.

Josverl, 2023-04-01:
I would be happy to help Thonny integrate the micropython stubs , especially as some of the internal bootstrapping code of an early Thonny version helped me understand one of the steps in the overall process of creating stubs.

aivarannamaa, 2023-04-02:
Thanks! Actually the stubs are already in Thonny. I suspect that it is the `jedi` library acting up. I'll investigate.