thonny/thonny #2675: Started Creating and Maintaining Snap Package..

CapeCrusader321, 2023-02-27:
Hey there, i have started building and maintaining Thonny Snap at https://snapcraft.io/thonny-ide, please checkout and help me by reporting any issues at https://github.com/CapeCrusader321/Thonny-snap/issues.

CapeCrusader321, 2023-02-28:
Can i assume that at point in  future it will be declared an Official Package ?? @aivarannamaa

CapeCrusader321, 2023-02-28:
You can check all the YAML etc., i can also help you to maintain and build it if needed..

CapeCrusader321, 2023-02-28:
https://github.com/thonny/thonny/issues/2418[](url)

CapeCrusader321, 2023-02-28:
#https://github.com/thonny/thonny/issues/2418

CapeCrusader321, 2023-03-07:
Hello @aivarannamaa  not heard from  along time, are you busy ?? Please consider the request whenever possible, Thanks..

aivarannamaa, 2023-03-07:
Thanks and sorry for the late reply! I have been really busy indeed.

> Please consider the request whenever possible

Do you mean mentioning this installation option at Thonny's homepage or something else?

I now added `sudo snap install thonny-ide` under Linux downloads at https://thonny.org/.

CapeCrusader321, 2023-03-07:
> Thanks and sorry for the late reply! I have been really busy indeed.
> 
> > Please consider the request whenever possible
> 
> Do you mean mentioning this installation option at Thonny's homepage or something else?
> 
> I now added `sudo snap install thonny-ide` under Linux downloads at https://thonny.org/.

Thanks for taking out your time,yes, mentioning it at home page will raise more awareness about the snap package, it already has 122+ downloads in 7 days...

aivarannamaa, 2023-03-11:
> I also have a doubt whether thonny uses dbus or not ??

Not at the moment, but please see https://github.com/thonny/thonny/pull/2683

closed 2023-03-11 by aivarannamaa