thonny/thonny #2676: Can't install Thonny 4.0.2 with Python 3.11.2

smblackledge, 2023-02-27:
Is there another way to install Thonny 4.0.2?

Raspberry Pi 4B, Buster (back one release :-(    ), 8G

1) Installation doc says Python 3.10 or later, IIRC, so I got the latest, 3.11.2.  Don't believe the Thonny install script supports that release.

2) pi@smb-raspi:~/Downloads $ ./thonny-4.0.2.bash 
Thonny only provides pre-built bundles for x86_64 (not armv7l).
pi@smb-raspi:~/Downloads $

Nothing happened.

smblackledge, 2023-02-27:
Downloaded [thonny-4.0.2-x86_64.tar.gz] but instructions don't mention what to do with it. Obviously untar it, but is there anything after that step?

CapeCrusader321, 2023-02-28:
It does have instructions, please check this link --> https://github.com/thonny/thonny/wiki/Linux

CapeCrusader321, 2023-02-28:
I am also developing a Snap package but its currently unstable and wont work properly, hope i will make it stable in the coming days...

smblackledge, 2023-03-01:
@CapeCrusader321, thanks for your info.  What I want is, indeed, there.

As an aside, don't think I didn't look.  I googled for things like "raspberry pi thonny 4 install", but all I got was Github pages and the Thonny web page (not the wiki page).

I know how difficult keeping information in one place. But it would be nice for install instructions to link to one page or to have all the information included in every place that Linux installation is described.

smblackledge, 2023-03-01:
And if I was lucky to find 4.0.4 because that version hasn't been released, then please let me know. Perhaps all mention of an unreleased version shouldn't be google-able.

smblackledge, 2023-03-01:
Removed strike-outs in OP.

smblackledge, 2023-03-01:
Looks like the Thonny installer really doesn't allow Python3.10. I wrongly believed that the supported version were everything 3.8 and after.

pi@smb-raspi:~/Downloads/thonny $ ./install
Copying files to /home/pi/apps/thonny ................................ Done!
Creating start menu item (/home/pi/.local/share/applications/Thonny.desktop)  Done!
Creating Desktop shortcut ............................................ Done!
Creating uninstaller (/home/pi/apps/thonny/bin/uninstall) ............ Done!
Compiling Python files ............................................... 
[Errno 8] Exec format error: '/home/pi/apps/thonny/bin/python3.10'
pi@smb-raspi:~/Downloads/thonny $

smblackledge, 2023-03-01:
Looks like the Thonny installer really doesn't allow Python3.11. I wrongly believed that the supported version were everything 3.8 and after.

```
pi@smb-raspi:/Downloads/thonny $ ./install
Copying files to /home/pi/apps/thonny ................................ Done!
Creating start menu item (/home/pi/.local/share/applications/Thonny.desktop) Done!
Creating Desktop shortcut ............................................ Done!
Creating uninstaller (/home/pi/apps/thonny/bin/uninstall) ............ Done!
Compiling Python files ...............................................
[Errno 8] Exec format error: '/home/pi/apps/thonny/bin/python3.10'
pi@smb-raspi:/Downloads/thonny $
```

CapeCrusader321, 2023-03-07:
As per Thonny homepage using  '''bash <(wget -O - https://thonny.org/installer-for-linux)'''  for linux lets you uses your system installed python version, so please try again by purging existing thonny completely and retry this method, i hope you get desired results else this should be addressed by the developers ??

brianlinuxing, 2023-03-10:
Just been doing some quick testing and I can confirm that Thonny works perfectly well on an x86_64 implementation of Debian Sid (SparkyLinux 7 (Orion-Belt)), kernel 6.1.0-5-amd64 on Python 3.11.2 (see screenshot)

I will try and perform more work on Thonny under Raspberry Pi OS and Diet Pi on a Pi 400 in the next few weeks and hopefully blog post on it, time permitting.

I shall also cover how best to install Thonny on various Linux distros (there are many possible ways), including on Raspberry Pi systems.
[image]

aivarannamaa, 2023-03-12:
@smblackledge, actually Thonny should be installable on Python 3.11. I've simply forgotten update the bash script. You can do it yourself by locating `GOODVER="Python 3\.(8|9|10)\.*"` and adding `11` as another alternative.

closed 2023-03-12 by aivarannamaa

amathenge, 2023-11-11:
> As per Thonny homepage using **_bash <(wget -O - https://thonny.org/installer-for-linux)_** for linux lets you uses your system installed python version, so please try again by purging existing [...]

This method worked for me. It created a new venv for Python 3.11 on my Pi and a desktop launch item.