thonny/thonny #2678: OpenCV via Manage plug-ins option not accessable

Petros626, 2023-02-28:
Hey there,

I've seen that several people had the problem to install OpenCV and use it in the _Thonny_ IDE. So I just installed it by searching `opencv-python` atfer that I restarted the IDE, but 
```python
import cv2
```
doesn't help, I get still `ModuleNotFoundError: No module named 'cv2'`. What could be the reason for it? In the terminal the module is also not recognized by `python3`.

Thanks in advance

aivarannamaa, 2023-02-28:
Did you use "Tools => Manage packages" or "Tools => Manage plug-ins" ?

Petros626, 2023-02-28:
I wrongly used `Manage plug-ins` and not the right one M`anage packages` . By the way whats the difference? Plug-ins is for the installed dependencies of a package?

Petros626, 2023-02-28:
So now I tried just to `import cv2 ` and the library (`picamera2`)  I'm using fails. Outcommented it's no problem, but I need OpenCV. The error message says the following:

[image]

Petros626, 2023-03-10:
I solved it with `sudo pip install -y python3-opencv opencv-data`. In case you install OpenCV from source, use this helpful discussion (https://github.com/Qengineering/Install-OpenCV-Raspberry-Pi-64-bits/issues/3)

closed 2023-03-10 by Petros626