thonny/thonny #2682: MemoryError: memory allocation failed, allocating 1979 bytes

Socialclimber, 2023-03-03:
This issue comes up when i try to import paho in the editor. funny enough the import works well when done on the shell.
stacktrace
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
MemoryError: memory allocation failed, allocating 1979 bytes

aivarannamaa, 2023-03-11:
Thank you for the report!

Which exact package are you referring to? Is it for CPython or MicroPython?

> This issue comes up when i try to import paho in the editor

Do you mean you get this error when writing in the editor or when running the code in the editor?

Socialclimber, 2023-03-11:
I think the problem Happened because I was running Python code in
micro-Python mode in thonny. Another issue I am having is that my micro
Python is failing to connect to local mosquitto broker. MQTTClient fails to
connect all the time. I running Ubuntu 22 on VMware.

aivarannamaa, 2023-03-12:
If you installed a CPython package on MicroPython device and imported it, then running out of memory is expected.

I'm afraid I can't debug your mqtt problem.

closed 2023-03-12 by aivarannamaa