thonny/thonny #2688: Console prompt up while turtle program is still running

nicolasdespres, 2023-03-07:
Hi,

I have noticed that when running an interactive program like the following using turtle the console prompt appear whereas the program is still running.

```python
from turtle import onkeypress, listen

def when_Up_pressed():
    print('Up arrow pressed')

onkeypress(when_Up_pressed, 'Up')
listen()
```

It leads to output like this, which is a bit confusing for beginners.

[image]

I don't know if there is an easy fix for this, since, I guess, it depends on the degree of integration of the turtle canvas.

Best regards,

aivarannamaa, 2023-03-11:
I'm considering this as a feature, as you can continue commanding the turtle in the REPL after the script ends. If you don't need this kind of interactivity, then you can add `turtle.mainloop()` to the end of your script.

nicolasdespres, 2023-03-11:
Ok fair enough. Thx for your quick reply.

closed 2023-03-11 by nicolasdespres