thonny/thonny #2689: Add option to change font size in variable panel

nicolasdespres, 2023-03-07:
Hi,

I often demonstrate coding using a projector. I increase the font size of the editor and the console so that students sitting in the back of the classroom can read.

With Thonny 3.3.14: 
[image]

With master Thonny:
[image]

With the new version the font size is no longer increased in the variable panel.

Best regards and thanks again for making Thonny,

aivarannamaa, 2023-03-11:
Thanks for the suggestion!

I'll deal with this issue together with other upcoming UI changes. In the meanwhile the work-around is to increase the UI scaling factor (Tools => Options => General).

nicolasdespres, 2023-03-11:
Ok thx for the tips.