thonny/thonny #2692: Thonny with micropython for the pico not loading.

Ollix27, 2023-03-09:
Hello, i only installed the firmware, connected the pico to my laptop running zorin OS 16.2, i also couldn't install the firmware from thonny, if i did it would just crash. i only selected the raspberry pi interpreter from thonny and it crashed with this error log:
Traceback (most recent call last):
  File "/snap/thonny-ide/26/lib/python3.10/site-packages/thonny/workbench.py", line 516, in _start_runner
    self._runner.start()
  File "/snap/thonny-ide/26/lib/python3.10/site-packages/thonny/running.py", line 118, in start
    self.restart_backend(False, True)
  File "/snap/thonny-ide/26/lib/python3.10/site-packages/thonny/running.py", line 686, in restart_backend
    self._proxy = backend_class(clean)
  File "/snap/thonny-ide/26/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 92, in __init__
    self._fix_port()
  File "/snap/thonny-ide/26/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 115, in _fix_port
    potential = self._detect_potential_ports()
  File "/snap/thonny-ide/26/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 200, in _detect_potential_ports
    all_ports = list_serial_ports()
  File "/snap/thonny-ide/26/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 988, in list_serial_ports
    return comports()
  File "/snap/thonny-ide/26/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in comports
    for info in [SysFS(d) for d in devices]
  File "/snap/thonny-ide/26/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in <listcomp>
    for info in [SysFS(d) for d in devices]
  File "/snap/thonny-ide/26/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 48, in __init__
    num_if = int(self.read_line(self.usb_device_path, 'bNumInterfaces'))
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

Could you help?

Ollix27, 2023-03-09:
Also i am in the dialout group if anybody's asking.

aivarannamaa, 2023-03-12:
Please try whether the official distribution (`bash <(wget -O - https://thonny.org/installer-for-linux)`) works for you. If yes, then report this issue at https://github.com/CapeCrusader321/Thonny-snap/issues

CapeCrusader321, 2023-03-23:
Any updates on this one @Ollix27  ??

postema, 2023-04-10:
Have same problem with the SNAP version.
the original runs fine
Ubuntu 22.04 LTS
Thonny 4.0.2
Raspberry Pi Pico

`Traceback (most recent call last):
  File "/snap/thonny-ide/37/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/snap/thonny-ide/37/lib/python3.10/site-packages/thonny/workbench.py", line 844, in _post_backend_menu
    entries = backend.proxy_class.get_switcher_entries()
  File "/snap/thonny-ide/37/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 379, in get_switcher_entries
    for device, desc in cls._detect_potential_ports():
  File "/snap/thonny-ide/37/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 200, in _detect_potential_ports
    all_ports = list_serial_ports()
  File "/snap/thonny-ide/37/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 988, in list_serial_ports
    return comports()
  File "/snap/thonny-ide/37/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in comports
    for info in [SysFS(d) for d in devices]
  File "/snap/thonny-ide/37/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in <listcomp>
    for info in [SysFS(d) for d in devices]
  File "/snap/thonny-ide/37/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 48, in __init__
    num_if = int(self.read_line(self.usb_device_path, 'bNumInterfaces'))
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

`

CapeCrusader321, 2023-04-16:
You can also refer to this if needed --> https://snapcraft.io/docs/troubleshooting

aivarannamaa, 2023-04-26:
Transferred to https://github.com/CapeCrusader321/Thonny-Snap/issues/2

closed 2023-04-26 by aivarannamaa