thonny/thonny #2694: Error in backend

bocephasj, 2023-03-09:
ERROR   thonny.plugins.micropython.backend: PROBLEM WITH THONNY'S BACK-END:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 347, in _handle_normal_command
    response = handler(cmd)

bocephasj, 2023-03-09:
After complete system update still more errors - PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ProtocolError: Could not get raw-paste confirmation).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.

aivarannamaa, 2023-03-12:
Thank you for the report!

Which device are you using? Which Thonny version?

Can you reproduce this error and post the backend.log (Tools => Open Thonny data folder) right after getting this error?

bocephasj, 2023-03-13:
Thonny 4 on Raspian(RPi4) working with DiGi Xbee3 radio module mounted on Grove Dev board. this problem came up very sudden and continues to very buggy (comes and goes). Since upgrading/updating it seems to have gotten worse.
Here is the newest;

Traceback (most recent call last):
  File "/usr/lib/python3.9/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 528, in _cmd_interrupt_with_shortcut
    self._cmd_interrupt()
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 489, in _cmd_interrupt
    self._proxy.interrupt()
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/mp_front.py", line 186, in interrupt
    self._send_msg(ImmediateCommand("interrupt"))
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 1075, in _send_msg
    self._proc.stdin.write(serialize_message(msg) + "\n")
BrokenPipeError: [Errno 32] Broken pipe

aivarannamaa, 2023-03-13:
This is now a different error. You got this after doing a Ctrl-C, right?

Can you give me the exact steps for reproducing the problem?

bocephasj, 2023-03-21:
I wish I could! Problem seems to be very inconsistent. some days it works without error, other times it does not want to work at all.

bocephasj, 2023-03-21:
This is a short program output-------------------
Firmware version: 0x1010
Hardware version: 0x414e
The XBee is 31.0 C (87.8 F)
Configured node identifier: co2-1
 +---------------------------------------+
 | XBee MicroPython Transmit Data Sample |
 +---------------------------------------+

Sending data to 0013A20041C812F9 >> Firmware version: 0x1010     Hardware version: 0x414e     The XBee is 31.0 C (87.8 F)     Configured node identifier: co2-1
Data sent successfully
-----------------------------------------------------------

After the program completes this is what happens 

PROBLEM IN THONNY'S BACK-END: Exception while handling 'Run' (thonny.plugins.micropython.mp_back.ProtocolError: Could not get raw-paste confirmation).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ProtocolError: Could not get raw-paste confirmation).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ProtocolError: Could not get raw-paste confirmation).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ProtocolError: Could not get raw-paste confirmation).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ProtocolError: Could not get raw-paste confirmation).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ProtocolError: Could not get raw-paste confirmation).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ProtocolError: Could not get raw-paste confirmation).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
MicroPython v1.12-1678-g32d849d on 2022-08-02; XBee3 Zigbee with EFR32MG
Type "help()" for more information.
>>> 
This is one of the shorter runs of the error returns that I get.

aivarannamaa, 2023-03-26:
I see that you have rather old MicroPython version. If it is not possible to upgrade it, then you could read https://github.com/thonny/thonny/wiki/MicroPython#advanced-configuration and try setting the `submit_mode` for your backend to `paste` or `raw`.

lovette-stack, 2023-04-05:
PROBLEM IN THONNY'S BACK-END: mainloop error (ValueError: bad marshal data (invalid reference)).
See Thonny's backend.log for more info.

Process ended with exit code 17.

lovette-stack, 2023-04-05:
The run feature and debuggers have been disabled