thonny/thonny #2695: Problem with python package: keyboard

Petros626, 2023-03-10:
Hey,

I can't run my script in Thonny or through the run in terminal option, because the system is blocking. I changed the mode for the whole folder with all permissions `sudo chmod -R 777`, but it still doesn't work. Its only with this specific package (https://pypi.org/project/keyboard/)

[image]

The suggestions I followed in several other forums don't help me.

aivarannamaa, 2023-03-12:
It looks like you need to start the process as a superuser. Thonny does not allow this. You need to run your script on the command line with `sudo`, eg. `sudo python3 my_script.py`

closed 2023-03-12 by aivarannamaa

Petros626, 2023-03-13:
The way to start in it terminal with `sudo` is one option. Is the following way not an option:
1. right click Thonny IDE
2. properties
3. desktop content
4. command 
5. paste in `sudo /usr/bin/thonny `