thonny/thonny #2697: Outline pane ignores multi-line strings

rauferd, 2023-03-10:
The outline pane currently doesn't recognize multi-line strings. Consider the following situation:
```
usefullString = """This is a description of what this program does. It prints this string!
 class UselessClass is not even syntacitally a correct class and
 def uselessFunction(first, second):
     print("at least looks like a function definition")"""

class SemiUsefullClass:
    def __init__(self):
        print(usefullString)

uselessInstance = SemiUsefullClass()
```
In this case `uselessClass` and `uselessFunction` show up in the outline pane, while they should not (in my opinion).

*Possible solution:*

It would be easy to check for """ and skip the parsing until the next occurance of """.

However, the following approach would be a better solution (in my opinion):
`def _parse_source(self, source):` in `outline.py` should use a tokenizer instead of a simple single-line-matching of `r"[ \t]*(async[ \t]+)?(?P<type>(def|class){1})[ ]+(?P<name>[\w]+)`

P.S.: I suppose it might be easy to fix #385 along the way.

P.P.S.: Do you want me to try and fix this (can't promise that I won't break something, though 😉)? If so, which tokenizer should I use?

aivarannamaa, 2023-03-12:
Thank you for the report!

You're right -- currently the Outline view does not outline this code properly, because the underlying implementation is really simple text search. 

You're welcome to propose a better implementation. The main problem is that a simple parser or tokenizer would choke on incomplete tokens (eg. multiline string without the final quotes). This means you need to choose between not updating the outline while tokenizing fails or accepting that several items outline items may disappear or re-appear because of single keypress.

You could try [parso](https://parso.readthedocs.io/en/latest/), which is meant for parsing incomplete code and is already included as transient dependency in Thonny. Not sure it is fast enough for parsing after each keypress, so you may need to introduce delayed updates, just like here: https://github.com/thonny/thonny/blob/1dc2ba52b754a6182fab83d1be0057455151598f/thonny/plugins/coloring.py#L153

 Or you could simply accept the extra items :)

rauferd, 2023-03-13:
Thank you for the quick reply and hints.

I created a very simple patch/pull request that fixes most common occurances of multi-line strings, including my example above and the typical "strings-as-comments" situation.

I will look into a more sophisticated solution including parso and `schedule_update` in the next couple of days/weeks.

For the time being, this works for me.

Feel free to close the issue or keep it open until a fully valid, 100% working fix is developed.

rauferd, 2023-11-15:
So far, I have not encountered problems with any "real life code" using this workaround in a couple of months.

From my point of view, this issue is fixed, the pull request will not be updated anymore and I abandoned the idea of programming a "real parser". Feel free to integrate this in next Thonny version.

If anyone is interested in programming a more sohpisticated parser, I am still willing to assist.

PR #2703 (2023-11-18): Update to outline.py to skip multi-line strings
Makes the outline "parser" skip multi-line strings. It works for all, but the most obscure cases, and (mostly) fixes #2697
commit 05a69b12 (2023-11-18): Update outline.py (#2703)

Makes the outline "parser" skip multi-line strings.
It works for all, but the most obscure cases, and (mostly) fixes #2697
closed 2023-11-18 by aivarannamaa