thonny/thonny #2698: Thonny error  when initializing backend

tonbe22, 2023-03-10:
Traceback (most recent call last):
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 516, in _start_runner
    self._runner.start()
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 118, in start
    self.restart_backend(False, True)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 686, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython_frontend\cp_front.py", line 36, in __init__
    super().__init__(clean, executable)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 950, in __init__
    self._start_background_process(clean=clean)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1053, in _start_background_process
    env=self._get_environment(),
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 993, in _get_environment
    env = get_environment_for_python_subprocess(self._mgmt_executable)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1395, in get_environment_for_python_subprocess
    overrides = get_environment_overrides_for_python_subprocess(target_executable)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1441, in get_environment_overrides_for_python_subprocess
    if os.path.samefile(
  File "C:\Users\User\AppData\Local\Programs\Thonny\.\lib\genericpath.py", line 101, in samefile
    s2 = os.stat(f2)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'Ä3瞈6膀莂薄螆覈變趌辎醐鎒閔鞖馘鮚鶜龞ꆠꎢꖤꞦꦨꮪ궬꾮놰뎲떴랶릸뮺붼뾾쇀쏂엄쟆질쯊췌쿎퇐폒헔ퟖ\ud9d8\U001069dc\udfde\ue1e0\ue3e2\ue5e4\ue7e6\ue9e8\uebea\uedec\uefee\uf1f0\uf3f2\uf5f4\uf7f6笠ﯺ﷼\ufffe'

aivarannamaa, 2023-03-12:
Thank you for the report!

I've seen similar reports (#2013), but I haven't been able to reproduce these. 

Which Windows version are you using? In which language?

It could be helpful if you could sent me your Thonny's configuration.ini (Tools => Open Thonny data folder).

tonbe22, 2023-03-16:
i use windows 7, 

[general]
configuration_creation_timestamp = 2023-03-10T22:53:48.896322
language = en_US
environment = []
single_instance = True
event_logging = False
disable_notification_sound = False
debug_mode = False
ui_mode = regular
scaling = default
font_scaling_mode = default

[view]
full_screen = False
maximize_view = False
ui_theme = Windows
shellview.visible = True
assistantview.visible = False
astview.visible = False
stackview.visible = False
exceptionview.visible = False
filesview.visible = False
heapview.visible = False
helpview.visible = False
notesview.visible = False
objectinspector.visible = False
outlineview.visible = False
todoview.visible = False
variablesview.visible = False
show_program_arguments = False
show_plotter = False
name_highlighting = True
locals_highlighting = False
paren_highlighting = True
syntax_coloring = True
highlight_tabs = True
highlight_current_line = False
show_line_numbers = True
recommended_line_length = 0

[LocalCPython]
executable = C:/Users/User/AppData/Local/Programs/Python/Python38/python.exe

[run]
backend_name = LocalCPython
program_arguments = 
dock_user_windows = False
pgzero_mode = False
auto_cd = True
birdseye_port = 7777
run_in_terminal_python_repl = False
run_in_terminal_keep_open = True

[assistance]
disabled_checks = []
open_assistant_on_errors = True
open_assistant_on_warnings = False
use_pylint = True
use_mypy = True

[layout]
zoomed = True
notebook_nw_visible_view = None
notebook_w_visible_view = None
notebook_sw_visible_view = None
notebook_s_visible_view = ShellView
notebook_ne_visible_view = None
notebook_e_visible_view = None
notebook_se_visible_view = None
west_pw_width = 165
east_pw_width = 165
nw_nb_height = 165
sw_nb_height = 165
s_nb_height = 165
se_nb_height = 165
ne_nb_height = 165
width = 800
height = 650
left = 883
top = 286

[file]
current_file = None
open_files = []
reopen_all_files = False
make_saved_shebang_scripts_executable = True
recent_files = ['D:\\Administrator\\Desktop\\Kurs uvod u programiranje\\udemy kurs\\Python Loops (Day 5 ).py']

[edit]
automatic_calltips = False
automatic_completions = False
automatic_completion_details = True
tab_request_completions_in_editors = False
tab_request_completions_in_shell = True
indent_with_tabs = False

[debugger]
frames_in_separate_windows = True
automatic_stack_view = True
allow_stepping_into_libraries = False
preferred_debugger = nicer

[shell]
clear_for_new_process = True
tty_mode = True
max_lines = 1000
squeeze_threshold = 1000
auto_inspect_values = True

tonbe22, 2023-03-20:
any chance for help?

tonbe22, 2023-03-21:
i did this and got  this in shell when i open again 
Interpreter 'ÄV齘Xईଊഌ༎ᄐጒᔔ\u1716ᤘᬚᴜ\u1f1e℠⌢┤✦⤨⬪\u2d2c⼮\u3130㌲㔴㜶㤸㬺㴼㼾䅀䍂䕄䝆䥈䭊䵌低児卒啔坖奘孚嵜彞慠换敤杦楨歪浬潮煰獲畴睶祸筺絼罾膀莂薄螆覈變趌辎醐鎒閔鞖馘鮚鶜龞ꆠꎢꖤꞦꦨꮪ궬꾮놰뎲떴랶릸뮺붼뾾쇀쏂엄쟆질쯊췌쿎퇐폒헔ퟖ\ud9d8\U001069dc\udfde\ue1e0\ue3e2\ue5e4\ue7e6\ue9e8\uebea\uedec\uefee\uf1f0\uf3f2\uf5f4\uf7f6笠ﯺ﷼\ufffe' not found.
Please select another!
Process ended with exit code None.

tonbe22, 2023-03-22:
yes
[image]
 but i got same problem when i do this like before

JChristensen, 2023-03-23:
FYI, I had this issue with Thonny 4.0.1 on a Ryzen 7 3700X desktop machine running Linux Mint 21.1, but not on an older laptop with a 3rd gen Core i7 also running LM21.1. Likewise no issue on a Raspberry Pi 4. Updating to Thonny 4.0.2 on the desktop machine fixed the issue for me.

aivarannamaa, 2023-03-23:
@JChristensen do you mean you have seen this `Interpreter <garbage text> not found` on Linux?

tonbe22, 2023-03-23:
yeah but i have windows 7 so 4.0.2  is not option for me :((

aivarannamaa, 2023-03-23:
https://github.com/thonny/thonny/releases/download/v4.0.2/thonny-py38-4.0.2.exe is meant for Windows 7

kilgore63, 2023-11-03:
I just installed Thonny via the UI / Snap on a fresh Ubuntu install on an HP laptop;  I am getting similar errors, and deleting the configuration file had no effect.

Any idea?

Traceback (most recent call last):
  File "/snap/thonny/210/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/snap/thonny/210/lib/python3.10/site-packages/thonny/workbench.py", line 847, in _post_backend_menu
    entries = backend.proxy_class.get_switcher_entries()
  File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 374, in get_switcher_entries
    for device, desc in cls._detect_potential_ports():
  File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 199, in _detect_potential_ports
    all_ports = list_serial_ports()
  File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 989, in list_serial_ports
    _PORTS_CACHE = _list_serial_ports_uncached(skip_logging=skip_logging)
  File "/snap/thonny/210/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 1023, in _list_serial_ports_uncached
    for p in comports():
  File "/snap/thonny/210/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in comports
    for info in [SysFS(d) for d in devices]
  File "/snap/thonny/210/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in <listcomp>
    for info in [SysFS(d) for d in devices]
  File "/snap/thonny/210/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 48, in __init__
    num_if = int(self.read_line(self.usb_device_path, 'bNumInterfaces'))
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

aivarannamaa, 2023-11-03:
@kilgore63, actually, your error is quite different from the one discussed in this issue. Better investigate https://github.com/thonny/thonny/issues?q=is%3Aissue+is%3Aopen+list_ports_linux.py and add your case under one of these issues