thonny/thonny #2700: Script containing PySide2 does not work when loaded from a file, but works if run from <untitled> page

AshDini, 2023-03-12:
[image]
[image]

aivarannamaa, 2023-03-12:
Please add following two lines in both programs and see if they print different value:

```python
import os
print(os.getcwd())
```

AshDini, 2023-03-13:
Hi Aivar,

Thank you for your prompt reply.

Something funny is going on here.

After receiving your email, I loaded the script from the file and tried it
out before inserting your suggestion. *And it worked perfectly !*

Since yesterday (when the problem occurred) I have not even loaded Thonny.
It seems that something happened yesterday with my several use of Thonny
that affected it.

However including your suggestion, it just printed the locality of the
file: "C:\ESP-Scripts\Keyboard" and continued working

Very strange !

Regards

Bob Stanton

closed 2023-03-13 by aivarannamaa