thonny/thonny #2701: How does thonny saves a file on the target device.

hacktronics, 2023-03-13:
This is a general question about how thonny works.
How does thonny saves a file on the target device.
I am currently using adafruit-ampy, but its too much hassle, wanted to know how thonny does the same.

aivarannamaa, 2023-03-13:
Thonny first tries to save the file via device's REPL using a set of MicroPython / CircuitPython statements. If this fails with "read only filesystem" error (as it is the case with default CircuitPython setup), then it tries to locate the mounted volume representing the device's filesystem and then saves the file via this mount.

hacktronics, 2023-03-13:
You mean executing something like below statements in the repl environment.
```
import os
f = open('/flash/main.py', 'w')
f.write('while True:\n')
f.write('  pyb.LED(3).toggle()\n')
f.write('  pyb.delay(1000)\n')
f.close()
os.sync()
```

hacktronics, 2023-03-13:
Also what does it send before those, CTRL+D or something else.

aivarannamaa, 2023-03-13:
Yes, that's basically how Thonny writes a file. Before file operations it doesn't do anything special. It does Ctrl-D in raw mode before running a file. See https://groups.google.com/g/thonny/c/z78gZQbVOs4/m/lay3HfeaBAAJ for more.