thonny/thonny #2702: Variable panel and breakpoint not working within a turtle callback.

nicolasdespres, 2023-03-13:
Hello,

With Thonny version 9a8fe782882721, variable panel is not updated (see `CLICKS_COUNT` in the panel) and breakpoint is not triggered within a turtle callback.

[image]

It fails with the embedded version or 3.10.2.

Cheers,

aivarannamaa, 2023-03-13:
Variables are only updated after the main thread returns (ie. on getting the new prompt in the shell) and when the debugger makes a stop. The click handler is just a function among many others and it is not feasible to update the variables after each function call.

aivarannamaa, 2023-03-13:
Oh, sorry ... I replied to quickly and didn't notice you are using the debugger. I'll investigate this.

nicolasdespres, 2023-03-13:
Indeed it is part of the answer. The title was misleading and was not mentioning that I was using the debugger. So the main issue is probably the breakpoint not being triggered. Thx for having a look.

aivarannamaa, 2023-03-13:
I can now explain this -- the debug session ended after you got the new `>>>` in the Shell. The reason you are still getting the click-events is that in the background Thonny keeps driving the message loop of the Turtle window. This is a hack inspired by IDLE.

Please try the same experiment after you have added `turtle.mainloop()` as the last statement in your program.

nicolasdespres, 2023-03-13:
The use of `turtle.mainloop()` does fix the #2688 but the breakpoint is still not triggered and now the variable panel is empty which is consistent with the fact that the main process did not return (fastdebug).

[image]

When using normal debug, the breakpoint does trigger:

[image]

and then control goes back to `mainloop` where we can still interact with the turtle canvas. It is a bit misleading since the debugger suggest the program is paused, but it works.

[image]
 

Here a copy-paste(able) version of the code if needed:
```python
from turtle import Screen, mainloop

CLICKS_COUNT = 0

def on_click(x, y):
    global CLICKS_COUNT
    CLICKS_COUNT = CLICKS_COUNT + 1
    print('Click #' + str(CLICKS_COUNT))

screen = Screen()
screen.onclick(on_click)
mainloop()
```

aivarannamaa, 2023-06-02:
I think it's because `global`-statement doesn't produce an instruction in the bytecode, ie. there is no instruction with line 6 and therefore there is no chance to stop on line 6.

I need to find a way to let user know about such ineffective breakpoints.