thonny/thonny #2704: Package install fails on webrepl connected esp8266

mediocre-to, 2023-03-14:
Hi. I am not sure if this is an issue woth considerating. If not, please close it immediately.

I tried to install umqtt and google to https://github.com/thonny/thonny/issues/2538 which is the reason I'd like to mention what happened.

Thonny was used on macos and connected by webrepl (esp in Wifi station mode) to the esp8266 connected with USB/Serial to a Windows box.

After an uninstall the esp was left in a state where the packagemanager stated the package to be installed (visible in the list on the left) but not usable by >>>import.

After the first install of umqtt all succeding tries to add/remove packages failed.

I used the os library and removed the package manually /lib/*.*. Still no luck.

Removing /lib gave the opportunity to install again, but only one time. I tested the procedure to manually remove and install with three different packages. Same result.

Moving on to the Windows box with the esp connected on serial. Installing and removing several packages worked fine, as expected

Thonny 4.0.2

MicroPython v1.19.1

The most siginificant error message: (ManagementError(...), 'Command output was not empty')

As a side note webrepl didn't work when Thonny was active on the Winbox. I assume that is "normal" but I had some troble understandig the error message (EOF reached ..... disconnected from webrepl ) 

Best regards Torsten

aivarannamaa, 2023-03-14:
Thank you for the report!

By `umqtt` did you mean [`micropython-umqtt.simple`](https://pypi.org/project/micropython-umqtt.simple/) ?

aivarannamaa, 2023-03-15:
Did I understand properly, that you had both serial and webrepl connection to the same device at the same time? I believe this doesn't work very well.

mediocre-to, 2023-03-23:
Sorry for late respons. 

No, serial and webrepl was not used att the same time to the same device.

The board was connected to a winbox but without thonny running. No other apps using serial to the board on that box meaning the winbox was effectively used as a powersupply. (Hopefully Windows didn't do anything in some background process.) 

Thonny using webrepl to that board was run on a macos laptop.

mediocre-to, 2023-07-18:
Ok. Nothing more!