thonny/thonny #2709: Confusing warnings about "Bad indentation"

amodia-bioservice, 2023-03-16:
After choosing 2 spaces as indentiation in your source files (which is "legal"!) AND in case of an error, Thonny throws warnings about this like:

Line 10 : Bad indentation. Found 2 spaces, expected 4

If an indentation of 2 (or any other number not equal to 4) is found AND consistent, this should not be warned at all. With the actual behaviour the output gets flooded with these useless warnings, "overgrowing" the more important warnings.
If you like to warn at all, then a single (!) "Unusual indentation of 2 spaces (standard: 4)" may be better.

aivarannamaa, 2023-03-16:
Please see https://github.com/thonny/thonny/wiki/Assistant

ethantse, 2023-03-29:
`print('your problem')`

KubaO, 2023-05-18:
Assistant works according to pep-8. If you don't use pep-8 indentation, then you'll need to disable indentation checking. IMHO this is a feature, not a bug. Python code indented by 2 spaces is a sign of trouble - usually caused by poor factoring. If the screen is "too narrow" with default indentation, then it's not the screen that's the problem :)