thonny/thonny #2710: Problem running Thonny

FredrikAndersson, 2023-03-16:
I get an error. I have uninstalled Thonny and reinstalled but the error persists.
Grateful for help:-) fredrik

PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.

I use Microsoft Windows 11 LATEST

aivarannamaa, 2023-03-17:
Thank you for the report! (Similar to #2585, #2520, #2513)

Please check, whether you can use your device with a simpler tool (e.g. Putty). If not then it's worth trying a more general MicroPython forum.

fwt-dot-fr, 2023-03-30:
exactly same issue:
putty serial  COM connected with out problem.

MartiMan79, 2024-08-22:
Same issue here, for me Thonny is NOT functioning anymore without flashing my pico W and reinstall.
Press stop once, bricked again...
Pico W is publishing and subscribing to MQTT perfectly but Thonny can't connect.

```
11:43:49.426 [MainThread] INFO    thonny: Thonny version: 4.1.4
11:43:49.426 [MainThread] INFO    thonny: cwd: D:\Documents\PICO\Stepper with home\FINAL
11:43:49.426 [MainThread] INFO    thonny: original argv: ['C:\\Users\\m_sme\\AppData\\Local\\Programs\\Thonny\\python.exe', '-s', '-u', '-B', 'C:\\Users\\m_sme\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\rp2040\\rp2040_back.py', "{'clean': False, 'port': 'COM4', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
11:43:49.426 [MainThread] INFO    thonny: sys.executable: C:\Users\m_sme\AppData\Local\Programs\Thonny\python.exe
11:43:49.426 [MainThread] INFO    thonny: sys.argv: ['C:\\Users\\m_sme\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\rp2040\\rp2040_back.py', "{'clean': False, 'port': 'COM4', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
11:43:49.426 [MainThread] INFO    thonny: sys.path: ['C:\\Users\\m_sme\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\rp2040', 'C:\\Users\\m_sme\\AppData\\Local\\Programs\\Thonny\\python310.zip', 'C:\\Users\\m_sme\\AppData\\Local\\Programs\\Thonny\\DLLs', 'C:\\Users\\m_sme\\AppData\\Local\\Programs\\Thonny\\lib', 'C:\\Users\\m_sme\\AppData\\Local\\Programs\\Thonny', 'C:\\Users\\m_sme\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages']
11:43:49.426 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=1, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
11:43:49.426 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': False, 'port': 'COM4', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
11:43:49.426 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 64, write_block_delay: 0, read_block_size: 0
11:43:49.426 [MainThread] INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type RP2040Backend
11:43:49.426 [MainThread] INFO    thonny: TIME/MODS 0.030 before prepare (+73 modules)
11:43:49.432 [MainThread] INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_ctypes', '_datetime', '_heapq', '_lzma', '_queue', '_struct', '_tkinter', 'binascii', 'bz2', 'ctypes', 'ctypes._endian', 'ctypes.wintypes', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'msvcrt', 'pathlib', 'platform', 'posixpath', 'queue', 'serial', 'serial.serialutil', 'serial.serialwin32', 'serial.win32', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
11:43:49.432 [MainThread] INFO    thonny.plugins.micropython.mp_back: Sending interrupt
11:43:49.432 [MainThread] INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
11:43:49.496 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Poking
11:43:49.496 [MainThread] ERROR   thonny.plugins.micropython.mp_back: Exception in MicroPython main method
Traceback (most recent call last):
  File "C:\Users\m_sme\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\mp_back.py", line 138, in __init__
    self._process_until_initial_prompt(
  File "C:\Users\m_sme\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 278, in _process_until_initial_prompt
    self._process_output_until_active_prompt(
  File "C:\Users\m_sme\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 861, in _process_output_until_active_prompt
    self._write(RAW_MODE_CMD)
[... cut ...]
  File "C:\Users\m_sme\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\serial_connection.py", line 88, in write
    size = self._serial.write(data)
  File "C:\Users\m_sme\AppData\Local\Programs\Thonny\lib\site-packages\serial\serialwin32.py", line 317, in write
    raise SerialException("WriteFile failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

```