thonny/thonny #2716: Installation fails on macOS

OliverGmelch, 2023-03-22:
I have just experienced a case where the Thonny installer on macOS 13.2.1 repeatedly failed with the error message "archive open failed".

Please see the attached log for more details.

Environment:
macOS 13.2.1
MacBookPro17,1

[Installationsprotokoll 22-Mar-2023.txt](https://github.com/thonny/thonny/files/11039540/Installationsprotokoll.22-Mar-2023.txt)

Any support would be highly appreciated.

OliverGmelch, 2023-03-22:
Yes, we have tried that, but unfortunately to no avail :/

aivarannamaa, 2023-03-26:
I suppose you need to do some [googling](https://www.google.com/search?q=%22archive+open+failed%22+macos+pkg&rlz=1C1CHBD_enEE867EE867&oq=%22archive+open+failed%22+macos+pkg&aqs=chrome..69i57j33i160l2.4198j0j7&sourceid=chrome&ie=UTF-8)

I found these:

* https://forums.macrumors.com/threads/install-with-pkg-file-failed.2367425/
* https://github.com/python/cpython/issues/100024 (where did you save the pkg?)

OliverGmelch, 2023-03-30:
Thank you so much, that did the trick, kudos to your Google Fu 👍 

Simple workaround: moved the installation package from the Downloads folder to the Desktop, et voilà, installation worked like a charm!

closed 2023-04-25 by aivarannamaa

kcarnold, 2023-08-16:
I hit this on Ventura 13.5, which is odd because it was supposedly fixed by Apple in 13.4.

Definitely the same problem: Dropbox is syncing the Downloads folder, and a system policy denied the installer from accessing `~/Library/CloudStorage/Dropbox/Downloads`.

The resolution of moving to Desktop won't work if Dropbox is also syncing Desktop. This should work:

1. Open new Finder window.
2. Go->Home (Command-Shift-H)
3. Drag and drop the installer there